armel/uv-k5-firmware-custom #215: feature request: Retain a single VFO

ct1esj, 2024-07-28:
Well, going from two vfo`s to just one, when you turn the radio off and on again, it always starts in double vfo.
Would it be possible to retain the last setting before shutting down?

Thanks for all the work and effort spent on the firmware, it's fantastic.
CT1ESJ - Henrique

prokrypt, 2024-07-28:
Use the menu to change rx mode to main only, then set your shortcut to change rx mode.
The shortcut functions are only for temporary settings.

armel, 2024-07-28:
Hello Henrique,

The display change is temporary. To make it permanent, use the RxMode menu (54/64).

Nevertheless, I understand your request. 

That's why I've added the MAIN ONLY action. Most of the time I use double VFO (MAIN TX DUAL RX). But I've added a MAIN ONLY action to the F1 (Short) button. So I can quickly switch to single display without having to go into the menus. 

73'

Armel F4HWN

ct1esj, 2024-07-29:
Yes, well seen, I hadn't made this combination of menu and keys, so it doesn't need to take up more space in the firmware.
Thank you Armel and Prokryt, for helping me find the switch, sometimes that's all it takes to make light.

armel, 2024-07-29:
You're welcome ;)

closed 2024-07-29 by armel