armel/uv-k5-firmware-custom #217: feature request: TX-Lock custom range and/or PMR 446 + Freenet

joernschmidt93, 2024-07-29:
Hi,

It would be nice to have the possibility to customize the TX-lock range in chirp or maybe use one of the channel-mode scan-list as a TX-whitelist.

If this feature isn't possible, it would be great to have a selection for PMR446 + Freenet in the hidden menu.

In germany there is not only PMR 446 but also Freenet (2m-Band) as a license free frequency range and it would be nice to use it, without interrupting other frequencies by accident.

Channel 1 | 149,0250 MHz
Channel 2 | 149,0375 MHz
Channel 3 | 149,0500 MHz
Channel 4 | 149,0875 MHz
Channel 5 | 149,1000 MHz
Channel 6 | 149,1125 MHz

armel, 2024-07-30:
Hi,

This is a good question. Well, could you take a look to my last survey on my [Youtube](https://www.youtube.com/@f4hwn/community) channel ?

Armel F4HWN.

joernschmidt93, 2024-07-30:
Hi Armel,

thanks for the quick response :)

just read your survey on youtube, the solution you suggested sounds great, if you can make it work this way, it would be just what i want.

I really appreciate what you are doing here

Joern

armel, 2024-07-31:
Hi Joern,

Thank you. I'll work on this for the next release. So stay tuned :)

Armel F4HWN.

Titivi09, 2024-07-31:
100

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

spamsool, 2024-08-08:
I checked on 3.2 but I couldn't find the option to enable only TX on PMR Band. It would be great to add PMR Only so I can sometimes borrow the radio to my non-ham friend and not worry if he transmits on frequencies which he shouldn't :)

armel, 2024-08-08:
Let me explain again... 

In past, I added band plans to the F Lock menu to meet various requests: PMR, FRS/GMRS/MURS, etc. But adding new F Lock options always takes up a lot of memory: new options in the F Lock menu, storing frequencies (for the specialists, these are uint32_t each time, so very memory-consuming), etc.

Now, it must be acknowledged that it will be complicated, if not impossible, to offer band plans that could cover and meet all expectations. It's not feasible, as highlighted by the completely legitimate request regarding the addition of Freenet. There are too many variations from one country to another. Worse still, those who would want an LPD band plan would certainly want a PMR band plan as well, at the same time. And nothing has been planned to possibly combine multiple band plans from the F Lock menu. In short, to cut it short, F Lock has got to many huge limitations and is not scalable.

That's because I've removed the PMR and FRS/GMRS/MURS options. Right? Wait for it, and keep in mind that it saved me some memory space. In v3.2, which I've just released, this memory space has enabled me to add an extra channel parameter: TX Lock (just as we already have W/N, BusyC, Compnd, and other parameters).

By default, TX Lock would be set to ON. But, from a new menu (which I called TXLock), it's now possible to switch it to OFF. In that case, it's possible to force transmission, rather than having TX DISABLE, even if the frequency is outside the band plan set from the F Lock menu. Of course, if a frequency is within the band plan set from the F Lock menu and TX Lock is ON, transmission is fully possible. The F Lock band plan take priority here.

Here is a practical example based on users who only operate on the PMR band. I know there are many. Until v3.2, they activated the PMR option from the F Lock menu.

With the new modification, they just need to choose DISABLE ALL from F Lock. In short, lock everything in TX. This helps avoid mistakes.

Then, from Chirp or directly from the radio, they just need to set the TX Lock parameter to OFF on the 16 PMR memory channels to create an exception and allow transmission. The advantage is that if they also want to operate on the LPD band (433 MHz), they could also do so on the concerned channels. The same goes for Freenet, etc. In short, you have the keys, you decide...

I've posted a [video](https://www.youtube.com/watch?v=ZdN9FlEwlOg) explaining the principle... Activate subtitles if needed. 

Armel F4HWN.

closed 2024-08-08 by armel

spamsool, 2024-08-08:
Thanks for the explanation, I've just tried and can't see the possibility to change TX Lock parameter to OFF in Chirp and driver 3.1
Will it become available in nex version of the driver?