armel/uv-k5-firmware-custom #219: Feature Proposal: Automatic Priority Channel for Scanning

zylka5, 2024-08-01:
Overview:
This proposal suggests adding an automatic priority channel function to the scanning feature in the Quansheng UV-K5 custom firmware.
Functionality:
The proposed feature would work as follows:

Initially, when scanning begins, there is no priority channel set.
Upon receiving a transmission on any channel:

This channel becomes Priority Channel 1 for a duration of one minute.
It will be scanned more frequently, out of the regular sequence of other channels.

If a transmission is received on another channel:

The current Priority Channel 1 becomes Priority Channel 2.
The new channel becomes Priority Channel 1.

Priority status is time-based:

After one minute without activity, a channel loses its Priority 1 status.
After two minutes without activity, a channel loses all priority status.

If no transmissions are received for two minutes, the scanning reverts to normal mode with no priority channels.

armel, 2024-08-01:
There is already a notion of priority channel in this firmware, easily configurable from Chirp. This is different from your suggestion, but it exist. 

[image]

closed 2024-08-03 by armel

zylka5, 2024-08-06:
Yes, I'm aware that the priority channel feature exists, but it requires access to a computer to change the parameters. When in the field in a dynamic situation, it would be beneficial to have an extension of this functionality accessible from the radio's menu.
This proposed feature would allow users to dynamically adapt to changing communication needs without requiring pre-configuration via Chirp. It would enhance the radio's flexibility and usability in situations where quick adjustments are necessary.

armel, 2024-08-07:
About your last request, simply use SList (20/65) to set your default preferred scan list. Mine is list [1], not [0].