armel/uv-k5-firmware-custom #221: No CW support?

londresarthur, 2024-08-02:
The information is not very clear: does this firmware support CW? In this case, pure CW, without modulation, just the carrier.

londresarthur, 2024-08-02:
edit1: it would be a very interesting implementation to do along with lower bandwidth options

armel, 2024-08-02:
Ok. This firmware does not support CW and there are no plans to add it.

londresarthur, 2024-08-02:
OK thanks. And is there manual gain control implemented? I really like this function in IJV, but Fabrizio doesn't seem to want to open source it so I can understand how he implemented it.

armel, 2024-08-02:
Manual gain control is possible only from the [Spectrum analyzer](https://github.com/armel/uv-k5-firmware-custom/wiki/Spectrum-analyzer#spectrum-sweep-screen).

Midnight-Cowboy, 2024-08-03:
Hi Armel, 
Is it possible to move the LNA settings outside the spectrum analyzer?

armel, 2024-08-03:
Hi @Midnight-Cowboy,

It's possible but with many modifications. I'm not sure I want to do it.

Midnight-Cowboy, 2024-08-03:
Maybe put this in a separate module and sacrifice something when compiling it yourself? For example, I don’t understand why the rx/tx timer is needed, and I could easily do without them.

armel, 2024-08-03:
@Midnight-Cowboy RX/TX timers are useful when using certain radio networks with TOTs. Second, it's not a memory issue... as I said, moving the LNA settings out of the spectrum analyzer is a huge challenge. And I'm not sure I want to do it. And above all, I don't have the time right now.

Midnight-Cowboy, 2024-08-03:
it's a pity...

armel, 2024-08-03:
@Midnight-Cowboy No, it's not a pity. My firmware is full open source. You can try to make this mofication and share it with the community if you wish 👍🏼 I'm sure other people will be interested. But I can't say yes to everything 😌 I volunteer on a whole range of other projects. And my time is limited 😔

Midnight-Cowboy, 2024-08-03:
I can solder, but I don't know how to program....

londresarthur, 2024-08-04:
And I can program, but like Armel, I'm totally out of time at the moment. Ahh how much easier life would be if Fabrizio was part of the open source community.

londresarthur, 2024-08-04:
Issue closed, maybe one day we will come together to implement new things. 73!

closed 2024-08-04 by londresarthur

wmarkow, 2024-09-09:
``` The information is not very clear: does this firmware support CW? In this case, pure CW, without modulation, just the carrier.

I think you can decode **CW** with **USB** demodulation mode. Just need to set the frequency i.e. around 700Hz below the frequency where **CW** is being transmitted, then you will hear a 700Hz tone of **CW**. 

If it goes about transmitting CW I have no idea; pressing PTT will generate carrier any way. You can check.