armel/uv-k5-firmware-custom #223: Evolution de SList

F6GDN, 2024-08-04:
Bonjour Armel,
A nouveau un grand merci pour l'énorme travail fourni pour le développement de la version 3 du firmware.
Les possibilités que tu ajoutes à chaque version sont épatantes.
J'ai une demande concernant le paramètre 19 du menu, soit SList.
Serait-il possible d'obtenir LIST [1,2], LIST [1,3] et LIST [2,3] dans ce paramètre ?
Au plaisir de lire ton commentaire à ce sujet.
Tous mes encouragements pour la suite !
Cordiales 73 !
Eddie
F6GDN

armel, 2024-08-04:
Bonsoir Eddie,

Merci pour ton message.
Et bien c'est intéressant, mais ca pose tout de même question.

Je résume ce qui fonctionne aujourd'hui: 

- si tu choisis LIST [0], ca va scanner les canaux présents dans aucune liste (donc 0)
- si tu choisis LIST [1], ca va scanner les canaux AU MOINS présents dans la liste 1 (donc 1, 12, 13, 123)
- si tu choisis LIST [2], ca va scanner les canaux AU MOINS présents dans la liste 2 (donc 2, 12, 23, 123)
- si tu choisis LIST [3], ca va scanner les canaux AU MOINS présents dans la liste 3 (donc 3, 13, 23, 123)
- si tu choisis LIST [1, 2, 3], ca va scanner les canaux AU MOINS présents dans une des listes 1, 2 OU 3 (donc 1, 2, 3, 12, 13, 23, 123)
- si tu choisis ALL, ca va scanner tous les canaux  (donc 0, 1, 12, 13, 23, 123)

Dès lors, que ce passe t-il si tu choisis LIST [1, 2] (par exemple) ? Ca ne va scanner que les canaux présents **AU MOINS dans les listes 1 OU 2** ? Si c'est le cas, ca concerne donc 1, 2, 12, 13, 23, 123 et cela revient presque à faire un scan LIST[1, 2, 3] comme actuellement, **sauf que les canaux présents dans 3 (uniquement) sont exclus**. 

Est-ce que le jeu en vaut la chandelle ?

J'aimerais bien avoir ton éclairage sur ce point.

Armel F4HWN.

F6GDN, 2024-08-05:
Bonjour Armel,

Merci pour ta réponse rapide et tes commentaires détaillés.

Dans mon esprit, la fonction SList LIST[n] scannait "uniquement", et pas
"au moins", les canaux mémoires enregistrés en liste "n".

Je m'étais donc organisé de la façon suivante :

- LIST [1] = les fréquences radioamateurs
- LIST [2] = les fréquences PMR
- LIST [3] = les fréquences aviation

De manière à choisir une écoute bien ciblée.

Au vu de tes explications, je vais réétudier la question et mon
organisation.

Merci au passage pour le driver CHIRP de la version 3.

A nouveau tous mes encouragements pour la suite.
Au plaisir.
Cordiales 73 !
Eddie
F6GDN

armel, 2024-08-05:
Bonjour Eddie,

Ton organisation est excellente et j'en utilise une similaire.

Mais effectivement, il faut aussi prendre en compte que l'on peut associer un canal à plusieurs listes. Et dans ce cas, la notion du "**au moins**" prend tout son sens.

C'est d'ailleurs pour cela que j'associe certains canaux aux listes [1, 2, 3], lorsqu'ils sont stratégiques. Par exemple si je reprends ton organisation:

- le canal 145.500 en VHF (fréquence d'appel RA) est sur la liste [1] (fréquences radioamateurs), mais je le positionne sur les 3 listes [1, 2, 3]
- le canal 121.500 VHF (fréquence de détresse aviation) est sur la liste [3] (fréquences aviation), mais je le positionne sur les 3 listes [1, 2, 3]
- etc.

Comme ca, même si je scan la liste 1 (fréquence radioamateur), si un appel de détresse est capté sur le 121.500 (fréquences aviation), j'ai une chance de l'entendre. Idem si je scan la liste 2 (fréquence PMR), je pourrais aussi capter les appels sur 145.500 en VHF (fréquence d'appel RA) ou sur 121.500 VHF (fréquence de détresse aviation). 

Ca permet de faire des combinaisons intéressantes (je crois).

73'

Armel F4HWN.

F6GDN, 2024-08-05:
Bonjour Armel ,

Merci pour ton message.
C'est exactement l'organisation des canaux "stratégiques" pour laquelle
j'ai optée avant même de recevoir ce nouveau message de ta part, en tenant
compte de tes observations précédentes !

Merci encore, et au plaisir de te suivre.

Cordiales 73 !
Eddie
F6GDN

armel, 2024-08-05:
Avec plaisir Eddie, merci à toi.
Je cloture cette issue :)

73'

Armel F4HWN.

closed 2024-08-05 by armel