armel/uv-k5-firmware-custom #227: How to renable scrambler?

OldMateAidos, 2024-08-10:
What option do I have to enable in the makefile to renable the scrambler

armel, 2024-08-10:
Sorry, but there's no parameter for that. And I don't plan to put such a parameter, in the Makefile, to allow it. I have no time to work on this. This feature is forbidden in many countries. I suggest you use a firmware like Egzumer if this feature is very important for you.

closed 2024-08-10 by armel

takuyozora, 2025-01-29:
Hi,

I've create a fork with a branch that add a compilation parameter to enable the scrambler : https://github.com/takuyozora/uv-k5-firmware-custom/tree/takuyozora/scramble

Just pass -DENABLE_SCRAMBLER to the make commande or use (or take example) the compile-with-docker.sh script with rescueops_scrambler 

Have fun