armel/uv-k5-firmware-custom #228: TXLock menu overrides FLock option

hypeARM, 2024-08-10:
If we are in VFO mode, even with FLock set to GB HAM, it is possible to transmit outside the permitted range due to the new TXLock option, which overrides the FLock setting

armel, 2024-08-10:
That's the idea. TXLock allows you to make exceptions on certain frequencies outside the band plan chosen via FLock. 
Take a look to the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus). So if in VFO mode you're able to transmit outside the permitted range (GB HAM for you), it's because FLock is OFF on your transceiver. 

Armel F4HWN.

zardoz99, 2024-08-10:
I seem to remember trying this out and I found that if you set TX Lock to ON on VFO A and VFO B, It works as expected and protects you from accidentally transmitting out of band. Buy default i have set everything to TX Lock ON and I'm using the Flock to decide whether transmit is permitted or not.

armel, 2024-08-10:
That's correct @zardoz99 :) Believe me, it's very simple :)

In memory channel or VFO mode, if the frequency is **inside** the band plan selected in FLock, you can transmit.
In memory channel or VFO mode, if the frequency is **outside** the band plan selected in FLock, 

- you can transmit **only** if TXLock is OFF
- you cannot transmit if TXLock is ON

Note that if a memory channel or VFO is outside the band plan and TXLock is ON, there will be a small padlock to the left of the name.

Armel F4HWN.

hypeARM, 2024-08-10:
For example, if I give TXLock ON within the GB HAM range all TX is blocked in VFO mode, for me it doesn't make sense to have TXLock ON, not allowing me to do TX within the GB HAM range.

hypeARM, 2024-08-10:
> I seem to remember trying this out and I found that if you set TX Lock to ON on VFO A and VFO B, It works as expected and protects you from accidentally transmitting out of band. By default I have set everything to TX Lock ON and I'm using the Flock to decide whether transmit is permitted or not.

It was just a mistake I made, that's what they said above, I just had to set TXLock outside of Range FLock in VFO mode to ON and everything was resolved!

closed 2024-08-10 by hypeARM