armel/uv-k5-firmware-custom #236: Txlock not working as expected

Linden1, 2024-08-21:
I expected Txlock to work with any channel so went through my channels locking any I didn't want to risk TXing on. This setting is indeed changed and saved to file, yet once uploaded to the radio the setting is lost for channels other than AM - specifically airband as I have no other AM channels to test with. I can set some up if you need? The setting is again lost for my FM channels when changed on the actual radio as well.

armel, 2024-08-21:
Trust me, it works.
What firmware version are you using? And what version of Chirp driver?

Linden1, 2024-08-21:
3.3 with 3.3.0 driver.
Is it right that this setting won't function if "UNLOCK ALL" is selected?

armel, 2024-08-21:
Yes of course.
**The FLock setting takes priority.** 
The right approach, if you're not a radio amateur, is to choose "DISABLE ALL" and then open exceptions on certain frequencies via TXLock.

Linden1, 2024-08-21:
I see. Opposite way to what I expected from my experience of these radios so far, but makes total sense.
Thank you.
Closing

closed 2024-08-21 by Linden1