armel/uv-k5-firmware-custom #241: Unable to use F Lock functions

cmpx, 2024-08-30:
Hi there.

I've upgraded to 3.3 Bandscope version from 3.2. 

I've set the F Lock Disable All bands for TX, and if I press PTT I'm told that TX is disabled. If I shut the device off to go back to normal mode without the hidden menu, I'm able to PTT and broadcast.  If I go back to menu mode, it shows that Disable All is still enabled but I can still broadcast in this mode (which I couldn't when I first set the setting). I am seeing the same results if I change the settings via Chirp, it takes the setting, but I'm then able to broadcast.

I've performed the Reset All and have also reflashed with the same results. I've also tried this by blocking only the FCC HAM bands with the same results.

Locking the channels individually works as expected.

armel, 2024-08-30:
And what about TXLock ? 
Take a look to the [Wiki ](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus)if necessary.

cmpx, 2024-08-30:
TX lock at the frequency or channel level works as expected.

armel, 2024-08-30:
If you have selected DISABLE ALL in the F-Lock menu and you can transmit, it's because the TXLock menu is OFF (on the active VFO or on the active memory channel).

cmpx, 2024-08-30:
Not sure I fully follow...

F Lock - DISABLE ALL - disables TX on all frequencies (get that)

TX Lock ON disables individual channels/VFO and OFF enables them. (get that)

The part I don't get is that I thought the behavior was always two separate use cases. F Lock to block out TX on bands or ALL. TX Lock for individual channels and frequencies. I don't recall one being dependent on the other.

Did F Lock  previously do the actual disabling without then needing the TX lock (per channel) based on what band options you selected? I haven't needed to change my Chirp channel files for months, and opening an old file from when I was on v 2.7 (maybe) I did not have all channels set to TX lock ON and I've had TX disabled since the beginning.

So, perhaps I missed a change in the functionality of the two somewhere along the line. Admittedly I hopped from 2.7 > 3.0 > 3.3 so maybe that's the case.

armel, 2024-08-30:
According to the changelog, FLock has been changed and TXLock added in version v3.2.

Before version 3.2, in memory channel or VFO mode, 

- if the frequency is inside the band plan selected in FLock (or if you have chosen to UNLOCK ALL), you can transmit.
- if the frequency is outside the band plan selected in FLock (or if you have chosen to DISABLE ALL), you cannot transmit. 

Since version 3.2, in memory channel or VFO mode, 

- if the frequency is inside the band plan selected in FLock (or if you have chosen to UNLOCK ALL), you can transmit.
- if the frequency is outside the band plan selected in FLock (or if you have chosen to DISABLE ALL),
   - you can transmit if TXLock is OFF
   - you cannot transmit if TXLock is ON

Armel F4HWN

cmpx, 2024-08-30:
Thank you for the clarification, I clearly missed this change as I skipped 3.1 and 3.2 coming from 3.0. Since I only use the DISABLE ALL option, it wasn't a consideration that it was changed.

I appreciate the work you've done, thanks again.

armel, 2024-08-30:
You're welcome. 

Armel F4HWN

closed 2024-08-30 by armel