armel/uv-k5-firmware-custom #243: Bug ou pas???

f1ixx, 2024-08-31:
Bonjour Armel

En bricolant dans le code (sauvegarde de data), j'ai vu ca.
Je n'ai aucun idee si ca peut etre un probleme.

J'allume le poste et fait un dump :
<pre>boris:[uv-k5-firmware-custom.original]$ k5prog -R 0x1FF0
Quansheng UV-K5 EEPROM programmer v0.4 (c) 2023 Jacek Lipkowski <sq5bpf@lipkowski.org>

******  Connected to firmware version: [F4HWN v3.3]
         00 01 02 03 04 05 06 07  08 09 0A 0B 0C 0D 0E 0F
0x1FF0   28 33 01 ff 01 8a 00 00  ff ff ff ff ff ff ff ff </pre>
Je fais la modif dans setTmr (c'est pas au pif ;-) : passe de <code>ON</code> a <code>OFF</code> ou l'inverse.
Re dump :
<pre>boris:[uv-k5-firmware-custom.original]$ k5prog -R 0x1FF0
Quansheng UV-K5 EEPROM programmer v0.4 (c) 2023 Jacek Lipkowski <sq5bpf@lipkowski.org>

******  Connected to firmware version: [F4HWN v3.3]
         00 01 02 03 04 05 06 07  08 09 0A 0B 0C 0D 0E 0F
0x1FF0   ff ff ff ff 00 8a 00 00  ff ff ff ff ff ff ff ff </pre>
hop, on pert les options de compilation en 0x1FF0 et 0x1FF1.

Mais ils reviennent lorsqu'on reboot le talky.
En faisant le poste, je n'ai pas regarde quand la sauvegarde du reglage sur spectrum est faite.
Je sais que c'est sur le Data[3]

Je ne sais pas si ca pose probleme pour d'autre fonction. Mais pour etre sur, j'ai modifie : 
setting.c - Ligne 720
original
<pre>#ifdef ENABLE_FEAT_F4HWN
    memset(State, 0xFF, sizeof(State));</pre>
changer en : 
<pre>#ifdef ENABLE_FEAT_F4HWN
    //memset(State, 0xFF, sizeof(State));
    EEPROM_ReadBuffer(0x1FF0, State, sizeof(State));</pre>

Grand merci pour l'enorme travaille et les explications.
73's

Boris

armel, 2024-08-31:
Bonsoir Boris,

Ca n'a aucune importance :) 
Les données sauvegardées aux adresses 0x1FF0 et 0x1FF1 sont réécrites à chaque démarrages du poste.

Voir main.c:103:    SETTINGS_WriteBuildOptions();
Tu trouveras la fonction SETTINGS_WriteBuildOptions(); dans settings.c:892

Au passage, l'intérêt de sauvegarder ces données me semble limité.
Il n'est pas impossible que je supprime cet appel.
En tout cas, je le ferais évoluer en stockant plutôt le type de firmware (bandscope, broadcast ou vox).

73'

Armel F4HWN.

f1ixx, 2024-09-01:
Bonjour Armel

Je viens de regarde dans le driver de Chirp.
Le pilote regard ces 2 variable pour definir les listes d'options.

Oui, re-ecrit a chaque demarrage, je pense donc qu'il est preferable de rebooter le talky a
chaque utilisation de chirp.

Bonne journee

Boris

armel, 2024-09-01:
Bonjour Boris,

En fait, il y a un gros héritage lié au firmware Egzumer ici. 

Jocelyn avait déjà travaillé sur le driver Egzumer et c'est donc naturellement en repartant de ce driver qu'il a travaillé sur le driver de mon firmware. Il y a donc un passif.

Certaines options de compilation peuvent influencer, par exemple, le choix des actions que l'on peut affecter aux touches F1, F2, etc. (exemple typique avec 1750Hz).

Mais pour moi, il y aurait du refactoring à faire ici. Et comme souvent, des modifications coté firmware auraient des répercutions coté driver, comme il y en a depuis le début de ce projet. Ce serait donc un travail à faire de concert avec Jocelyn, comme à chaque fois ;)

Au delà de 0x1FF0 et 0x1FF1, il y a d'autres octets "mal utilisés" pour sauvegarder les settings dans l'eeprom :) Et là, ca date d'avant Egzumer.

73'

Armel F4HWN.

f1ixx, 2024-09-02:
Bonjour

Oui, je pense bien qu'il y'a encore pas mal de travail sur ses parties.
Je continus a regarder les evolutions du le code que vous faite.
C'est amusant (pour moi)  d'essayer de comprend ce que fait le
bouillons d'octets dans le talky. 

Merci pour le retour et les infos
Grand Merci a Jocelyn pour le super boulot sur le driver Chirp.

Boris

closed 2024-09-29 by f1ixx