armel/uv-k5-firmware-custom #249: F Lock setting is not honored

bbirbo, 2024-09-13:
After upgrading to v3.3 the F Lock is not honored and even it is set to DISABLE ALL it still allows the radio to TX.

Also when this menu is set from CHIRP the value that you set is not saved correctly on the radio.

armel, 2024-09-13:
First of all, hello to you...

Secondly, please, I think you should learn how to use this transciever by reading [online ](https://github.com/armel/uv-k5-firmware-custom/wiki)documentation. If you can still transmit, check the TX Lock settings (menu 13/66).

Read or read again this [section](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus).

Last but not least, about Chirp, if you think there is an issue, post it on [this repository](https://github.com/armel/uv-k5-chirp-driver). Thank you.

Armel F4HWN

bbirbo, 2024-09-13:
Thanks for the quick response but I still dont get it , according to the documentation the DISABLE ALL should work correct ?
Wouldn't that prevent TX for all channels ? Is there a way to set TX to OFF for all the channels ?

armel, 2024-09-13:
Once again, check whether TX lock is ON on your memory channels, either directly from your transceiver or from Chirp. If all your memory channels have TX lock set to ON and you set DISABLE ALL, you won't be able to transmit.

Armel F4HWN

closed 2024-09-14 by armel