armel/uv-k5-firmware-custom #251: Bug in displaying frequency in scanning mode in airband

suchyh, 2024-09-15:
Hello! 
First of all I want to thank you for the great firmware which is amazing and I like it a lot!

I have noticed that after upgrading to version 3.3, broadcast version, scanning in airband zone works fine but
the frequency showed on display have a bug. 
Starting at 107.99166 MHz the next step (8.33kHz) is showed on display 108.00000 and in next step is showed 
18.00000 next one is 18.00833.
I haven't tested any further.

https://github.com/user-attachments/assets/b4b2d597-2ae8-41ce-9b39-6e345f1f6c76

armel, 2024-09-15:
Hello,

Thank you for your message.

Ok, so, in VFO mode, the BK4819 (radio chip) uses several frequency ranges :
- 01800000 to 10800000
- 10800000 to 13700000
- 13700000 to 17400000
- 17400000 to 35000000
- 35000000 to 40000000
- 40000000 to 47000000
- 47000000 to 130000000

So if you enter the frequency 10799166, you're in the first range ( 01800000 to 10800000). And when you reach the high limit (10800000), you're back at the low limit (01800000).

Hope this help :) 

Armel F4HWN.

suchyh, 2024-09-16:
Ok, I get it.
Thank you!
I will close the issue!

closed 2024-09-16 by suchyh