armel/uv-k5-firmware-custom #253: Battery is always on 100% / 8.33V

HorstBaumann, 2024-09-18:
Everything works great, but on my Retevis RA79 (identical in construction to the Quansheng UV-K5),
the battery value does not change - its alway 100% and 8.33V.

I use the original Battery of the RA79, which has 1400mAh (7.4V 10.36Wh)

prokrypt, 2024-09-18:
Have you confirmed with a multimeter that the voltage does not change?
Have you calibrated the reading from within the radio?

HorstBaumann, 2024-09-19:
The multimeter says 8.09V - when I press PTT it goes down to 7.90V.
No change on the Display 8.33V.
I did no calibration on the radio yet.

armel, 2024-09-19:
And? Why don't you do the calibration like everyone else? This feature is there for a good reason, not for fun.

HorstBaumann, 2024-09-19:
sorry, but i didn't know where. 
in "BatType" (71/72), i can just select 1600, 2200 and 3500mAh.

OK - no i realised, that in "BatCal" (70/72), i can change the value - sorry.
Thanks for your fast response

closed 2024-09-23 by armel