armel/uv-k5-firmware-custom #255: Wide or narrow stuck

m6vmo, 2024-09-23:
Hi,
when I set W/N form WIDE to NAROOW, it works, but when I later turn the radio off and back on, it has gone back to WIDE.
Am I doing something wrong or is it a bug in the firmware (3.3).

I would like to set the radio to NAROOW permanently.

Thank you.

prokrypt, 2024-09-23:
You need to (re)save the channel for those settings to stick

armel, 2024-09-23:
Hello,

The reason is : in memory channel mode, W/N is a **"per channel" parameter**. You can change it from the radio. But if you want this change to persist (after changing channel, restarting the pocket, etc.), **you need to save the memory channel again (via menu 17 ChSave)**. Otherwise, the change is only temporary, as you've seen for yourself.

The same behavior applies to mode, power, W/N, BusyCL, ... and all “per channel” parameters. This is normal usage on all (commercial) transceivers.

Regards,

Armel F4HWN.

m6vmo, 2024-09-23:
Thank you ever so much.

That really helped me out.

This email has been checked for viruses by Avast antivirus software.
www.avast.com