armel/uv-k5-firmware-custom #262: Écran

F4JQC, 2024-09-30:
Slt Armel j'essaie d'avoir une deuxième fois le bug, mais j'ai eu un bug en sortie de sommeil du TX ... Il m'a mis l'écran en mode miroir, pour ce faire j'ai voulu sortir du sommeil en appuyant sur le PTT et hop j'ai eu cette affichage... 73 de 
F4JQC et merci pour ton travail
[image]

F4JQC, 2024-09-30:
J'ai essayé de le reproduire, mais impossible... Ca n'est peut-être pas du au firmware, mais uniquement a mon Quansheng.
Je verrai par la suite....

armel, 2024-09-30:
Bonjour François,

Ah oui, c'est amusant cet effet miroir sur l'écran. 
Je ne l'ai pas constaté de mon coté, après plusieurs jours de test (et de sortie de veille).
Mais un bug est toujours possible.
L'écran est-il resté comme cela, même après plusieurs appuis sur les touches ? 
Si tu arrives à le reproduire, tiens moi au courant. 

73'

Armel F4HWN.

armel, 2024-10-03:
C'est reproductible ou pas ?

giteetlecouvert, 2024-10-03:
Pour l'instant pas... je vais essayer de trouver ce qui influence ça...

F4JQC, 2024-10-03:
> C'est reproductible ou pas ?

J'ai essayé, mais ca ne me la pas refait ..... Mais ca ne gene pas le fonctionnement

giteetlecouvert, 2024-10-03:
bon j'ai réussi ce soir, à peu près une fois sur 5.
l'écran inverse est fonctionnel (F+A/B change réellement le triangle de
sélection), par contre tout appui sur PTT remet l'écran à l'endroit.

oui ce n'est pas gênant, mais c'est rigolo comme problème... et j'ai du mal
a voir d'où ca pourrait venir...

armel, 2024-10-04:
@giteetlecouvert Ok pour ce retour, merci. C'est compliqué de fixer un comportement qui n'est pas simplement reproductible. Ca le rend plus difficile à comprendre et à analyser. De mon coté, je n'arrive pas à avoir cet effet en sortant de veille par appui sur le PTT. Mais je tenterais à nouveau.

Remarque en complément, je trouve tout de même un peu bizarre de sortir de veille par appui sur le PTT. On ne sait même pas forcement sur quelle fréquence on est. Ca me semble un peu curieux comme approche :)

Armel F4HWN.

F4JQC, 2024-10-04:
> @giteetlecouvert Ok pour ce retour, merci. C'est compliqué de fixer un comportement qui n'est pas simplement reproductible. Ca le rend plus difficile à comprendre et à analyser. De mon coté, je [...]

Slt Armel,
Pour ma part, la sortie de veille sur le PTT c'est faite en saisissant le pocket a pleine main et comme j'ai des doigts de fée,  je n'avais pas fait attention.... Mais là, pourtant essayé plusieurs fois de refaire cet effet, mais sans succès.... (Bon j'y passe pas mes journées non plus) HI...

F4JQC, 2024-10-04:
A force de persévérance.... Les conditions (que je pense) hier pas d'allumage du pocket, ce matin 1er allumage, mis en veille (1mn) un coup de PTT (volontaire ce coup si) et hop écran miroir 🤣, mais aucune piste 😤
[image]

F4JQC, 2024-10-04:
Sur FB, d'autres commentaires sur (l'effet miroir)

armel, 2024-10-04:
@F4JQC oui vu, ca semble toujours être dans le cas d'un réveil suite à l'appui sur le PTT.

armel, 2024-10-05:
Bonjour à tous,

Après de très nombreux tests, je pense avoir fixé le problème, du moins je l'espère.
Il devrait donc être fixé dans la prochaine version v3.5 sur laquelle je développe encore.
Voilà pour les nouvelles et encore merci à @F4JQC d'avoir ouvert cette issue. 

Bon week-end,

Armel F4HWN.

armel, 2024-10-12:
Fixé dans la version v3.5 que je viens de publier. Merci à tous.

closed 2024-10-12 by armel