armel/uv-k5-firmware-custom #263: Frequency Unlock settings not being set

jimmusson, 2024-09-30:
Whatever I set the Tx Frequency Lock to it will let me transmit on any frequency.
Example: I set it to FCC HAM and I can transmit on 224.260 and 162.550.
I have tried the 3.3 and 3.4 firmware  versions using the 3.3 and 3.4_k chirp modules.  
Using Chirp dated 7/6/24 because the newest one bricks the ijv firmware.

jimmusson, 2024-09-30:
Quansheng UV-k5 (8)

armel, 2024-09-30:
First of all, hello.

What about the TXLock setting (menu 13/67) ?
If TXLock is OFF, what you see is normal.
Please read [this note](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus) on the Wiki.

73'

Armel F4HWN.

jimmusson, 2024-09-30:
Hi and thanks for responding.  I completely missed the Tx Lock setting.  Changed it on the radio and it now works as expected.
Is the Tx Lock in the chirp settings?

armel, 2024-09-30:
Yes, of course, for each memory channel. A column is dedicated for this. 

73'

Armel F4HWN.

jimmusson, 2024-09-30:
ok thanks, but no field equivalent to menu 13?

armel, 2024-09-30:
As I said, yes, of course, for each memory channel. A column is dedicated for this.
[image]

closed 2024-10-02 by armel