armel/uv-k5-firmware-custom #264: Deep Sleep not fully explained

awawaddfdf, 2024-10-02:
This might seam pretty obvious what it does, but on the wiki it just explains how to turn on deep sleep, but doesn't explain what it does.
Also been using your firmware for over a year

armel, 2024-10-02:
First of all, hello.

That is correct. Excuse me but :
- I explained how it works in my last video. Even if it's in French, it's still possible to activate the automatic translation (even if it's not perfect),
- developing this firmware takes up a lot of my time, and I have other priorities,
- I just haven't had the time to do it yet.

I'll take care of it when I have time. 

Armel F4HWN.

armel, 2024-10-02:
Here is the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-setoff-menu).

Armel F4HWN

closed 2024-10-02 by armel