armel/uv-k5-firmware-custom #265: PB affichage.

xavbo, 2024-10-04:
Bonjour.
Merci pour ce vous faites. Nous apprécions.

Avec la V3.4, lors d'une séquence de touche toute simple dans les menus de paramettrage, je me suis retrouvé avec l'écran inversé dans le sens droite gauche.
J'ai ré-appuyé sur une touche et il s'est remis à l'endroit.

Curieux, mais pas trop gênant.

Xavier

xavbo, 2024-10-04:
Sinon, existe t'il une notice expliquant les menus et les raccourcis clavier ?
Merci de votre retour.
Xavier

xavbo, 2024-10-04:
Il est possible que l'inversion d'écran se soit faite lorsque j'ai débranche la prose de programmation.

armel, 2024-10-04:
Bonsoir,

Pour la documentation, voir le [wiki](https://github.com/armel/uv-k5-firmware-custom/wiki).
pour le problème d'affichage, en l'état, il me faut plus de détails. Le problème est-il reproductible et si oui comment ?

Armel F4HWN.

armel, 2024-10-05:
Bug corrigé dans la version 3.5 en cours de développement. Je ferme ici.

closed 2024-10-05 by armel

xavbo, 2024-10-06:
Bonjour armelÇa y est. J’ai réussi à rendre reproductible le pb. Voici la séquence : Lorsque l’appareil est en veille avec la fonction setoff. Avec un autre tx on émet un cours instant de tel sorte à ne pas réveiller l’appareil en veille. Ensuite on appui sur le ppt de l’appareil en veille et là, l’écran est affiché à l’envers. Voici la photo. Pour remettre l’affichage de l’écran à l’endroit, il faut appuyer sur pptSinon. J’ai un pb avec en mode verrouillé Je ne trouve pas comment paramétrer le tx pour qu’en mode verrouillé le ppt soit toujours actif.  Je suis preneur d’infos. Par ailleurs. En mode verrouillé la touche f1 que j’ai paramètre en scan reste active. Ce qui pose des pb pour toujours être sur le canal du groupe avec lequel on communique. Je ne sais pas si je vous sollicite trop et donc vous fait perdre du temps ou si au contraire vous attendez ce type de retour. A votre dispo. Merci pour ce que vous faites. Xavier

Closed #265 as completed.

armel, 2024-10-06:
Bonjour,

Alors comme dit, ce bug a été corrigé dans la version 3.5 qui sera publié prochainement. Au passage, attention à vérifier qu'un bug n'a pas déjà été remonté. Concernant l'écran en miroir, [c'était le cas](https://github.com/armel/uv-k5-firmware-custom/issues/262). Ca permet d'éviter les doublons.

Pour le verrouillage du TX hors PTT, voir le menu SetLck (et choisir l'option Keys et non Keys + PTT).  

Tout est dans la documentation. Il faut la lire.

Armel F4HWN