armel/uv-k5-firmware-custom #266: Band plan

SortableTrash67, 2024-10-05:
I chose band plan unlock all and now I want to go back to default+. In the hidden menu, I changed Flock back to deafult+ and the lettering in the menu changes, but the radio still broadcasts on all frequencies. I tried to change the band plan a few times and even with disable all radio broadcasts on all frequencies, I also tried to change the firmware to a different one, but still the same. I uploaded the original firmware and then the range was locked, but after returning to this firmware, the radio is unlocked again

armel, 2024-10-05:
Hello,

You're in memory channel mode or frequency mode ?
What about TXLock menu (menu 13) : ON or OFF ?
Take a look to the [wiki](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus) if necessary.

Armel F4HWN.

SortableTrash67, 2024-10-05:
I read the wiki but I didn't fully understand how tx lock works. I was confused that in VFO and MR mode I can broadcast wherever I want outside the band plan, it turns out that everywhere I had TXLock off by default.
Thanks, once you set TXLock to it works as it should.

armel, 2024-10-05:
Ok, no problem. Thank you ;) 

Armel F4HWN

closed 2024-10-05 by armel