armel/uv-k5-firmware-custom #272: Egzumer flash tool no longer working on edge and chrome om macOS

stanclaes, 2024-10-11:
Egzumer flash tool no longer working on edge and chrome om macOS .

Loading file from url: https://github.com/armel/uv-k5-firmware-custom/raw/main/archive/f4hwn.bandscope.packed.v3.4.bin

CRC check passed...
Detected firmware version: *F4HWN v3.4
Firmware uses 99.14% of available memory (60908/61439 bytes).
Connecting to the serial port...
No data received, is the radio connected and in flash mode? Please try again.
Connecting to the serial port...
Failed to connect to the serial port.
[image]

armel, 2024-10-11:
Trust me, it works like a charm.
Probably an issue of USB driver.
I run dozens of flash from chrome on Macos (Sonoma 14.6.1) every day.

stanclaes, 2024-10-11:
YesIt worked for me fine until I upgraded my Macs to the latest Mac os version…Can you confirm it works fine on your side with latest Mac OS ?In the mean time my chirp is still happily connecting to the radios… that makes me assume it is not driver related… but, no expert here.Any suggestions welcome…Best RegardsStanVerstuurd vanaf mijn iPhone

armel, 2024-10-11:
Ok, interesting. Thank you for this report. 

Well, I haven't installed Macos Sequoia because I'm currently developing projects with Macos Sonoma for my work. Macos Sonoma is my target stack operating system. By the way "_if it works, don't fix it_" :) But if I do install Macos Sequoia, which I don't plan for the moment, I'll let you know. No problem. 

Finally, please note that I'm not developing Egzumer's flash tool. You could try to post an issue [here](https://github.com/egzumer/uvtools/issues).

Armel F4HWN.

stanclaes, 2024-10-11:
done, thx!

armel, 2024-10-11:
Hi,

I've just tried Egzumer flash tool on a Macbook Pro running Sequoia, with Chrome: it works.

Armel F4HWN.

stanclaes, 2024-10-12:
Ok, thanks for that feedback.... bizarre ... I have the problem on 3 different macs that each were working perfectly with the flash tool before upgrading...

ssombra, 2024-10-12:
Hi. That happened to me also. I was using the flashing tool without problems and suddenly it just stopped working; however chirp did work fine. I changed the cable and it is working again. I am not sure why the cable worked with chirp but not with the flashing tool. Maybe try either a different cable.

fachinformatiker, 2024-10-13:
I had this problem too.
MacBook Pro M1, newest MacOS update, Chrome
Three Radios aren't flashable anymore.
Tried it on Windows11, works like a charm

Chirp is working without problems on MacOS

armel, 2024-10-13:
Questions: 
- Chrome is up to date ?
- Have you tested with another cable ?

Armel F4HWN

stanclaes, 2024-10-13:
for me Chrome Version 129.0.6668.91 (Official Build) (arm64). nor alternative cables available unfortunately...

armel, 2024-10-13:
Ok guys.

So I tested on several machines (Macbook Pro Intel and Macbook Air M1) under Macos Sequoia 15.0.1. It works.
As far as your problem is concerned, I think it's a problem with the cable and/or the USB driver.

https://github.com/user-attachments/assets/add4b7e2-7a49-4465-8908-28716d47bb33

73'

Armel F4HWN.

ssombra, 2024-10-13:
> Ok guys.
>
>
>
> So I tested on several machines (Macbook Pro Intel and Macbook Air M1) under Macos Sequoia 15.0.1. It works.
>
> As far as your problem is concerned, I think it's a problem with the cable [...]

I agree. Changing the cable fixed the problem for me. What it is weird is that chirp continues working with the old cable.

armel, 2024-10-16:
Closed. This is not a problem about to the flash tool.

closed 2024-10-16 by armel