armel/uv-k5-firmware-custom #276: Should 4.2s countdown time be removed while in  MAIN TX & DUAL WATCH mode？

troilus, 2024-10-17:
Thanks for your firmware!
Should 4.2s countdown time be removed while in  MAIN TX & DUAL WATCH mode？Because in this mode user always TX on select channel/VFO, so it is not necessary to wait for 4.2s to resume dw mode and ensure TX on the RX channel/VFO.

armel, 2024-10-17:
Hi,

Sorry, what 4.2s countdown time are you talking about?

troilus, 2024-10-17:
In DW mode, after the radio receives a signal and the signal ends, the radio will remain on the VFO that just received the signal. According to the code, this time is 4.2 seconds. I’m thinking that in MAIN TX DUAL RX mode, this time doesn't seem to be very necessary.

https://github.com/armel/uv-k5-firmware-custom/blob/0a7e836656c1b66b8f4c2dc1995ab1600b397738/misc.c#L48

armel, 2024-10-17:
Ok. But where is the issue ? 
Of course, you're free to change it if you want 😌