armel/uv-k5-firmware-custom #277: Fix FM broadcast RX frequency range.

troilus, 2024-10-17:
Hi F4HWN,
    Recenty I want to fix the FM broadcast RX frequency range, so user can not Press F+1 to change the range while in FM radio.
I changed the lim from
```
uint16_t BK1080_GetFreqLoLimit(uint8_t band)
{
    uint16_t lim[] = {875, 760, 760, 640};
    return lim[band % 4];
}

uint16_t BK1080_GetFreqHiLimit(uint8_t band)
{
    band %= 4;
    uint16_t lim[] = {1080, 1080, 900, 760};
    return lim[band % 4];
}
```
to 
```
uint16_t BK1080_GetFreqLoLimit(uint8_t band)
{
    uint16_t lim[] = {870, 870, 870, 870};
    return lim[band % 4];
}

uint16_t BK1080_GetFreqHiLimit(uint8_t band)
{
    band %= 4;
    uint16_t lim[] = {1080, 1080, 1080, 1080};
    return lim[band % 4];
}
```
But seems the actual frequency received by the radio does not match the input frequency during actual use.

prokrypt, 2024-10-17:
Idea: only allow the range to be set when there are no frequencies stored in memory.

armel, 2024-10-17:
Sorry, but what is the use case ?
The RX frequency for FM broadcasting varies from country to country. That's why F+1 is used to select the desired range. And you want to impose 87MHz to 108MHz on everyone? Why would you do that?

troilus, 2024-10-17:
Sorry I am making a firmware for some chind based on your work, I tried press F+1 to switch the 4 band plan after edit, and input freq in VFO mode, still work wrong.

armel, 2024-10-17:
Trust me, F+1 switch works like a charm.

closed 2024-10-17 by armel