armel/uv-k5-firmware-custom #285: Spectrum with range mod

lucknum24, 2024-10-26:
When I used the range mod in spectrum function, the arrow is on correct location, but the spectrum length is wrong on scale. 
[图片]
Is that my problem?

lucknum24, 2024-10-26:
[image]

armel, 2024-10-27:
Hi,

Thank you for this issue.

Where is the problem ?
Can you tell me more?

[image]

Armel.

lucknum24, 2024-10-27:
[image]
[image]
[image]
@armel   Sorry, I can only speak a little English. Could you understand what I said?
Thank you for your reply.

DL2MF, 2024-10-27:
Hi Armel,

long time not seen ;-) I can confirm that I noticed also above issue. The marker ist at the correct position, but the spectrum above does not scale to the complete screen width (seems like scale factor calculation / lines per kHz) fails.

armel, 2024-10-27:
Hello,

Ok,
I see. I don't have a solution for the moment.
I tag this issue as a bug.

Armel.

armel, 2024-10-28:
@lucknum24 I'd like to ask you a question. How do you get a frequency step of x40 or x160?

lucknum24, 2024-10-28:
@armel 
[image]
Use the ScnRng mode. If use the normal mode , there are only 16x,32x,64x,128x bars number choices by pressing 4.  
But in ScnRng mode, the initial bars number is calculated by program and can not be changed by pressing 4. Changing the freq step by pressing 1 and 7, the bars number is also changing.
Did I said clearly? I can take a mp4 to describe it.

armel, 2024-10-28:
@lucknum24 Crystal clear, thank you 😉

armel, 2024-10-28:
@lucknum24 By the way, I can now reproduce the problem. That's a good thing. Thank you.

lucknum24, 2024-10-28:
@armel Thanks for your custom firmware. I think the problem is from the original author. I'v used egzumer edition, and some fork edition from egzumer. All have this problem. I hope you can fix it. Thank you very much!

lucknum24, 2024-10-28:
I remembered that the original author is fagci, who created the spectrum function.

armel, 2024-10-28:
@lucknum24 That's right. The spectrum analyzer comes from Fagci. It's a special firmware extension. A sort of application within an application. It has its own management of keyboard shortcuts, backlighting and so on. I've made very few modifications on it.

fagci, 2024-11-08:
Неправильно вы бутерброд едите, колбасой вниз надо.

@fagci <- вот так упомянуть можно, или сослаться в issue в основном репозитории

lucknum24, 2024-11-08:
@fagci Ok, I see. I apologize for that. Thank you for your correction.

armel, 2024-11-08:
@fagci Thank you for your intervention and, above all, thank you for your excellent work 🙏🏼

armel, 2024-11-08:
Fixed by @fagci in next release v3.7.