armel/uv-k5-firmware-custom #286: After long reception time scan doesn't resume

Linden1, 2024-10-27:
I've never seen this as it actually happens but if I leave my radio scanning a group, if it picks up a signal it'll pause scanning so I can listen. When I come back to the radio later scanning has stopped altogether and the keypad has locked. It seems like if scanning is paused for long enough the scan function fails to resume and the keypad locks (as it automatically should). If I disable the key lock scanning will again stop but without the key lock.
This is with the bandscope version of 3.5

armel, 2024-10-27:
First of all, Hi.

What settings do you have in ScanRev (24/67) ?
What settings do you have in KeyLck (30/67) ?

Armel.

Linden1, 2024-10-28:
ScanRev: 10s:000ms
KeyLock: 02m:00s

armel, 2024-10-28:
@Linden1 Ok, I will try to reproduce the problem.

Linden1, 2024-11-02:
@armel I can't reproduce it on another radio with the same settings. I think this radio is either on it's way out or the memory is somehow corrupted. Close this if you want.

armel, 2024-11-02:
@Linden1 Interesting stuff. I tried to reproduce the problem on my side too. But without success. It's strange 🤔

armel, 2024-11-06:
I'm closing this issue because I wasn't able to reproduce it. I will reopen it if new elements can help to reproduce and understand the problem.

commit 89415768 (2024-01-14): Fix #286: Lack of continuous tuning and monitor lockout while holding up/down button in monitor mode.
closed 2024-11-06 by armel