armel/uv-k5-firmware-custom #289: [enhancement] Disable menu button

emanuelegissi, 2024-11-03:
Hi, we are firefighters, we would like to lock down the radio, so that our less savvy colleagues cannot accidentally change any setting.

We can already set the stock `VFO open` to false.
I suggest adding an option to lock the menu as well, or at least hide it behind a combination of keys (eg. PTT + ... + M).

Thank you very much for your hard work. And thanks again in any case.

armel, 2024-11-03:
Hi,

Thank you for your message.
Keyboard lock is not sufficient (long press on F or in automatic mode via KeyLck menu) ?
When the keypad lock is active, it is not possible to enter Menu mode. 

Armel.

emanuelegissi, 2024-11-04:
Hi Armel,
Unfortunately keyboard lock does not allow switching channels.

The goal is using the K5 as a business two way handheld for firefighters. It is very cheap, good battery and usb-c, it has UHF band for ff local coms, VHF band for medical coms, marine channels, wildfire freqs, PMR for mountain rescue, and allow listening our 70MHz vehicular channels...

I could not find a better handheld radio for firefighters. An only radio for all the coms, a dream.

In an ideal world, single PTT, volume setting, channel switch, and large font channel name display would be all we need.

Emanuele

emanuelegissi, 2024-11-04:
J'ai oublié : merci beaucoup pour ta réponse rapide !

emanuelegissi, 2024-11-07:
Hi Armel, may I ask where in the code the menu button is managed? I could try to send you a pull request on this feature, for your evaluation.
Thank you

rob-w, 2024-11-09:
Hi 
like , Volume up/down on the side buttons ? 
What about having menu only when menu was pressed on turn on , otherwise blocked ?

armel, 2024-11-09:
It's an idea worth exploring.

rob-w, 2024-11-09:
i call it "talkie" mode :)   no F-Key function too, have the keyboard only set channel numbers :)

emanuelegissi, 2024-11-10:
That would be great! Thank you

emanuelegissi, 2024-11-12:
Armel, this sounds great!
I have a couple of deadlines hanging over me, then I will test your code in a controlled operational setting and keep you posted.
Merci beaucoup !

armel, 2024-11-12:
> Armel, this sounds great! I have a couple of deadlines hanging over me, then I will test your code in a controlled operational setting and keep you posted. Merci beaucoup !

No problem. Take your time, there's no hurry. Your feedback will be invaluable. We're going to develop this rescue version and improve it together. It's too important and I'm well aware of it.

Armel F4HWN.

armel, 2024-11-14:
@emanuelegissi 

New version [f4hwn.rescueops.packed.bin.zip](https://github.com/user-attachments/files/17742425/f4hwn.rescueops.packed.bin.zip).

Please note that, due to GitHub's security limitations, this is a zipped binary. You'll need to unzip it before using it. 

I've started writing a little [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#rescueops-edition) for this version. 

Your feedback will be invaluable. I know you have a busy schedule. Please take a look when you can. For my part, I plan to introduce this version for next version 3.7 (in addition to the Bandscope, Broadcast and Voxless versions). And I'll do my best to improve it based on feedback from emergency response professionals.

Regards,

Armel F4HWN.

armel, 2024-11-14:
🚨 First public version in last [release](https://github.com/armel/uv-k5-firmware-custom/releases/tag/v3.7) 😉

emanuelegissi, 2024-11-15:
You are amazing!
Still busy but intend to put the radios to real world test next week.

armel, 2024-11-15:
Hi,

Have you take a look to the [Quansheng Dock](https://github.com/nicsure/QuanshengDock) project ?

Armel F4HWN

armel, 2024-11-15:
Hi there,

@emanuelegissi Ok, keep me informed and we'll improve this version according to your ideas and needs. 

@lohtse Thank you for this report. Very interesting. In France, I think we use Tetrapol (Antares for the fire department). It's digital (encrypted). And transceiver costs are (obviously) very high. Do you see any areas for improvement in the current version of RescueOps?

Armel F4HWN.

emanuelegissi, 2024-11-16:
If you want to only see the name, set Channel Display Mode (ChDisp) to `NAME`.

emanuelegissi, 2024-11-16:
I uploaded the firmware to two UV-K5(99), and set all our channels.
Next Monday I am starting the operational test. This weekend I start playing.

As a first attempt, I set the F1 button to lock/unlock, and the F2 button to LED.

I will let you know. But the first impression is... amazing.

emanuelegissi, 2024-11-16:
> > If you want to only see the name, set Channel Display Mode (ChDisp) to `NAME`.
> 
> I know that lol...But having Frq there but in smaller text is very useful as explained.

armel, 2024-11-16:
Hello everyone ;)

@lohtse Humm, it will be hard. I use the largest alphabetical font currently available. Be aware that a channel name could use 10 characters. If I find an even larger font, it might cause problems beyond 6 or 7 characters. It could be a bit funky to change this... But I'll take a look if I have time.

@emanuelegissi F1 Lock and F2 LED seem to be a good configuration. I can't wait for the feedback in the field, in real condition. It will be very interesting. And we'll make the necessary adjustments. 

Thank you,

Armel F4HWN

armel, 2024-11-18:
@lohtse Hello. If you compile version 3.8 under development from the feature_update_v3 branch, you can now choose the Tiny display in Main Only mode (using the SetGui menu). So, when rescueops mode is activated, you'll be able to have a display like the one on yesterday's screenshots.

emanuelegissi, 2024-11-19:
Hello, after some days of testing, the RescueOps mode seems *very well accepted* by my firefighters as it is now. They add their thanks to your valuable work.

Anyway, here are a few suggestions I would like to share, if easily attainable:
 1. maintain the lock/unlock functionality of the `F#key` button; some of the ff see that key icon, and try to lock/unlock the radio with that button.
 2. allow temporarely setting the max power level with the `F1` or `F2` buttons.
 3. allow temporarely disabling the `frequency offset`  with the `F1` or `F2` buttons.

The second point needs further explanation. I program a default 1W power on the team com channels. That power is enough for local ops, and this helps saving the battery. But in certain cases (eg. searching for missing people in hostile environments) a power boost is needed. Would it be possible to press the `F1` button and have the transmitting power rise to the 5W maximum?

No need for intermediate levels for the sake of simplicity, and the channel programmed power should not be changed. If the ff switches off the radio, the boost can be forgotten.

The third point needs an explanation too. I have many channels with a repeater. But sometimes the direct communication is preferred by the teams, especially in areas that are badly covered by our repeaters.
I currently have to save the same frequency on two different channels (with and without frequency shift, repeated and direct). When the  the `F1` or `F2` button is pressed, the frequency shift is switched off and a label appears on the screen, eg. "DIRECT COM" or just a "D" after the channel name.

The channel programmed shift should not be changed. If the ff switches off the radio, the direct setting can be forgotten.

If this is too much work, don't worry. The radio is already good enough as it is now.
And if you have any better idea, it is very welcome of course.

Ps ff stays for firefighter!

armel, 2024-11-20:
@emanuelegissi Thank you for this beautiful report. It is very interesting. I'll see what I can do. Failure is not an option 😌

wen24o2, 2024-11-21:
Hi. RescueOps mode looks more like ChildLock Protection mode to me, but it is great. The only thing I miss is unlocking the mode with a PIN code.

Thank you for your work put into the development of the firmware. I always watch your videos demonstrating new features and fixes (with automatic translation because I don't speak French). You are the best.

armel, 2024-11-21:
@wen24o2 The difference between an usage intended for professionals and one intended for children lies in the fact that the former sees this radio as a work tool and aims to use it without taking the risk of changing anything, while the latter, on the other hand, specifically wants to alter the settings and bypass the restrictions. The target of this firmware is the first category, not the second. Adding a PIN code introduces extra complexity. I don't have the time to dedicate to it, sorry.

armel, 2024-11-25:
Hi @emanuelegissi, @lohtse and the others,

New version [f4hwn.rescueops.packed.bin.zip](https://github.com/user-attachments/files/17897066/f4hwn.rescueops.packed.bin.zip) under development (remember to unzip before flashing).
About changes:

- maintain the lock/unlock functionality of the F#key button; some of the ff see that key icon, and try to lock/unlock the radio with that button : **done**.
- allow temporarily setting the max power level with the F1 or F2 buttons : **done**, take a look to the new POWER HIGH action
- allow temporarily disabling the duplex (frequency shift) with the F1 or F2 buttons: **done**, take a look to the new REMOVE SHIFT action
- the Tiny style can now be applied to Main Only display, which makes frequency and channel name more visible. 

Thank you for your comments, when you have time to test.

Armel F4HWN.

emanuelegissi, 2024-11-25:
Hi Armel, this same morning (hopefully) I am going to update this testing firmware, and let you know.

Thank you, from the time of your post (si vous êtes en France) it seems that you worked all the night long.

armel, 2024-11-25:
@emanuelegissi Yes, I worked on it last night, but I'd made good progress during the week. I took advantage of the weekend to do some tests :)

emanuelegissi, 2024-11-30:
Hi Armel, these nights I did my homework finally.

I learned how to compile the feature branch and uploaded it to one radio. It worked!

I started testing the new features. Everything seems to behave as expected. The result is **great**. No other word to define it. It seems perfect.

The ffs love the RxMode set to MAIN TX / DUAL RX. And realized very quickly how to switch from MR A to MR B. So I think I am going to set it as default.

I just have a configuration doubt left.
When not in rescueops mode, the DUAL RX is shown like this:

[image]

But when in rescueops mode I don't find a way to show the same DUAL RX. Only the main RX is shown:

[image]

What am I missing? What config should I change to obtain the same visualization in the rescueops mode?

Thank you.

armel, 2024-11-30:
Hello @emanuelegissi,

Thank you for your message.

Ok Emanuele, no problem :) In a previous [message](https://github.com/armel/uv-k5-firmware-custom/issues/289#issuecomment-2454011373), you mentioned wanting a large font size for displaying the channel name. I had therefore decided that if the RescueOps mode was activated, it would force the display to switch to MAIN ONLY (single VFO, large font, etc.). However, I can remove this behavior. The display mode will remain as the one you selected before activating RescueOps mode (MAIN TX DUAL RX, DUAL RX RESPOND, CROSS BAND, or MAIN ONLY).

Here is a new [version](https://github.com/user-attachments/files/17963743/f4hwn.rescueops.packed.bin.zip) with the [fix](https://github.com/armel/uv-k5-firmware-custom/commit/4cb25739fc4076a7dd396231fb77c6d49520bc57).

If you prefer to compile it yourself, you can use the [feature_update_v3](https://github.com/armel/uv-k5-firmware-custom/tree/feature_update_v3) branch.

Regards and 73'

Armel F4HWN

armel, 2024-11-30:
@emanuelegissi I almost forgot, it’s still possible to assign shortcuts to the F1, F2, or M keys with the RX MODE action (which cycles through the different display modes) or the MAIN ONLY action (which switches directly to Main Only mode). These shortcuts remain functional in RescueOps mode :)

emanuelegissi, 2024-11-30:
@armel okay, I understand.

And looking at your commits I start learning where the behaviors are set. I may dare to submit pull requests in the future, but feel absolutely free to reject my crappy attempts...

Thank you again. It is like if you *gifted* us with new radios. I am now going to flash 10 radios, write some wiki documentation limited to rescue ops only, and start a new round of real ops tests.

Imho, the rescueops version is now *feature complete*. If you want, we could consider this issue as closed. I would open a new issue if any problem arises.

Ps If you ever decide to travel to Genova, Italy, you will be our guest!

armel, 2024-12-01:
@emanuelegissi Thanks for the nice feedback 🙏🏻 It was very interesting to work on this special version. I'm glad you like it 🤗 I'm going to close this issue. But don't hesitate to open a new one if you have other needs / ideas. And thanks for the invitation, that so kind of you 😌 Stay safe, you and your firefighting team, and thank you for saving lives.

closed 2024-12-01 by armel