armel/uv-k5-firmware-custom #293: V3.6 issue

AgyFater, 2024-11-06:
I don't have 72 menu item, only 70 with the new, 3.6 FW (excl. HIDDEN MENU). I uploaded Broadcast and Bandscope version. Is it possible to forget something? TIA.

armel, 2024-11-06:
First of all, Hi.

According to the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Menu#main-menu):

- MicBar - microphone bar that appears while transmitting (⛔️ not available with the Voxless version)
- VOX - voice TX activation sensitivity level VOX Setting (⛔️ not available with the Voxless version)

So it's normal to have only 70 menu item, not 72.

Armel F4HWN.

closed 2024-11-06 by armel

AgyFater, 2024-11-06:
Thank you for your answer. BTW I can't find 'NARROWER' option in menu. I think, its my fault. 😊

armel, 2024-11-06:
Don't search it, not enough space. Use broadcast OR bandscope version 😉