armel/uv-k5-firmware-custom #299: F4HWN V3.6 keyboard delay

BENGEE94, 2024-11-11:
Keyboard delay is too short so when inputting frequency into VFO for example: 027.555 will become 7--.---
Once you re input the frequency it will correct itself.

armel, 2024-11-11:
First of all, Hi...

Too short, really ?

https://github.com/user-attachments/assets/ec6a4fda-208d-4b35-be0a-44528474a1a0

BENGEE94, 2024-11-11:
Hi Armel thank you for such a speedy response, it is like a glitch more
than a delay so in the first input it will skip 02 and just input the 7,
this isn't every time but i never noticed it on previous version's of the
firmware

armel, 2024-11-11:
There has been an evolution in version v3.6. I've explained it in a video. Before v3.6, if you wanted to enter the frequency 145.500 MHz, you had to press 1 4 5 5 0 0. Now, with v3.6, you simply enter 1 4 5 5 and that's it. The firmware will take care of the rest and add zeros. In addition, the delay before keyboard input stops has been lowered. Before v3.6, it was 8 seconds (an eternity...). Now it's 1.5 seconds. I'll raise it to 2 seconds in the next v3.7 version.

BENGEE94, 2024-11-11:
Again thank you for the speedy response and I will look forward to 3.7 as I
really enjoy the firmware 👍

armel, 2024-11-14:
Improved in v3.7 😌

closed 2024-11-14 by armel

BENGEE94, 2024-11-14:
Thank you very much sir and I can't wait to give it a try