armel/uv-k5-firmware-custom #301: Inverted keys in BatCal menu?

rezemika, 2024-11-12:
Hello, thank you so much for your firmware and for all your work on it (and all its documentation and videos too)!

I'm not sure it's a real bug, maybe it's the intended behavior, but I guess the up/down keys are inverted in the BatCal menu (tested on a Quansheng K6 with F4HWN v3.6): when I press the *up* key, the displayed voltage is *decreased* (and vice-versa, the *down* key *increases* the displayed voltage). It's really not a big deal, I just wondered if it was intended. ^^

Have a nice day!

armel, 2024-11-12:
Hello 🤗

When you use the up and down keys, you increase (up key) or decrease (down key) the second value (UNDER battery voltage). It's the change in this second value that affects the voltage. 

Armel F4HWN.

prokrypt, 2024-11-13:
Just want to add: it's done this way because pressing a button changes the calibration value, but the voltage does not always change with it. Therefore it's easier to get feedback of proper operation from the calibration value (up to increase, down to decrease)

rezemika, 2024-11-13:
Oooh my bad, sorry for the issue, and thanks for the explanation! :)

closed 2024-11-13 by rezemika