armel/uv-k5-firmware-custom #308: Spectrum range issue in V3.7

DL2MF, 2024-11-16:
Hi Armel,

thanks for fixing the spectrum display issue with V3.7 and support by @fagci 

[image]

But as I have noticed, that the spectrum range labels still remain being faulty. In the attached example the edge frequencies are correct, but the bandwidth/range of +/- 800kHz does not fit.

73 Mike, DL2MF

armel, 2024-11-16:
Hi,

According to the document : the value that appears in the lower central part of the screen is to change the start/end frequency of the sweep, each time we press the up/down key, that is, if we have started from 118.4MHz and that value is, for example +-2000k, it means that when we press the up arrow, the initial sweep frequency will now be 2MHz higher, that is, 120.4MHz. If we press a second time, it will change to 122.4MHz and so on. This function is useful when we want to do a complete sweep in sections. 

You can change this value by pressing 2 or 8 key.

Armel F4HWN.

DL2MF, 2024-11-17:
Hi, 
ok mni tnx for this explanation. I didn't realize, that this is the spectrum change step size. The function of 1/7 and 2/8 keys were clear to me, but up/down key doesn't change anything in the sprectrum mode on my UV-K5. I'll check, waht is going wrong here.
Mike, DL2MF

closed 2024-11-17 by DL2MF