armel/uv-k5-firmware-custom #309: Channel name overlaps spectrum

DL2MF, 2024-11-16:
When using spectrum scan close to a strong signal, the (auto added) name and frequency overlaps the spectrum.

[image]

In the small gaps between the characters the reals spectrum bar height is visible.
Writing to the display should be reversed:
- first static information like frequency and name (I'd like to disable also the automatically addition of the channel name from memory)
- then the spectrum that may also overlap the other channel info

This might be more useful than to cut the spectrum on top.

73 Mike, DL2MF

armel, 2024-11-16:
Hi,

Again, according to the documentation, adjust the upper signal level of the spectrum, (the Y axis). In the upper left part of the screen we see two negative numbers, separated by a bar. The first is the minimum signal received, it is a value that will be around -130dBm and the second value, we can adjust it with the “3” and “9” keys until it is at an approximate value between -70 and -30dBm to our liking, depending on We want the spectrum to more or less “fill” the screen.

Armel F4HWN.

DL2MF, 2024-11-17:
OK, tnx! 
I'm using the spectrum display for fox hunting and radiosonde chasing. In this usecase it's much more comfortable not to reduce the dynamic range to about the half of the available screen size. The resolution in the target area is the much to low and the slow scan speed of the UV-K5 is also challenging. Reading the dB values might be an option, but is also not comfortable in the field.
But I understand, that my requirement may be very specific, so feel free to close this issue, if there is no way to realize this without big effort.
Best regards,
Mike, DL2MF

armel, 2024-12-13:
Hello Mike,

In the next version, the channel name will be displayed in the status bar. This will probably be better. 
Stay tuned.

Armel F4HWN.

DL2MF, 2024-12-20:
Hi Armel,
thanks for this enhancement. I've already made a custom V3.9 this week and it looks/work fine for me now ;-)
Mike, DL2MF

closed 2024-12-20 by armel