armel/uv-k5-firmware-custom #317: BatTXT value changes via Chirp do not write to radio

9of7, 2024-11-19:
Unsure if a firmware issue or Chirp issue:

Value changes to BatTXT (None / Voltage / Percent) in Chirp (Settings / Basic Settings) do not upload to the radio. Instead, the BatTXT default value (None) is always active after each Memories/Settings upload to the handheld. 

Radio: Quansheng UV-K5(8) (egzumer + f4hwn)
Chirp: v. next-20241115
Chirp Module: uvk5_egzumer_f4hwn_ver_3_7_0.py

armel, 2024-11-19:
First of all, Hi...

Thanks for this (probably old) bug. It's fixed in the [latest version](https://github.com/armel/uv-k5-chirp-driver/releases/tag/3.7.1) I've just published. Note that, for Chirp issues, it's best to use Issue [here](https://github.com/armel/uv-k5-chirp-driver/issues) :) 

Thanks again,

Armel F4HWN.

9of7, 2024-11-19:
Thank you, Armel.

Yes, the latest version of Chirp module, 3-7-1, resolved this issue.

armel, 2024-11-20:
Fixed un Chirp v3.7.1

closed 2024-11-20 by armel