armel/uv-k5-firmware-custom #322: Spectrum marker doesn't fit to frequency

DL2MF, 2024-11-22:
After the successfull bugfix of the spectrum with V3.7 it looks, like in some situation the frequency marker does not fit the detected and displayed maximum frequency.

[image]
Sample 1: 405.700 MHz is the maximun in the right half of the display, marker is in the left.

[image]
Sample 2: 405.300 MHz has been detected, marker should be near to the center, where the maximum bar is correct displayed

I noticed this on all my radios with FW3.7 and a friend of mine confirmed the same error on his radios. 

It looks, like the marker is being calculated depending on the old (smaller) buggy range in some cases. Sometimes the marker is being displayed with a delay of > 5 Sec. at the correct position.

You might have to change resolution/decimation and frequency range to reproduce this behaviour.

armel, 2024-11-23:
Hi,

Some news...

For the moment, I was not able to reproduce this issue. But I'm not saying it doesn't exist. 

I configured a first K5 under the same conditions, scanning from 405.1400 to 405.7800 with a 10 kHz step, resulting in 64 samples. With a second K5, I transmitted at very (very...) low power on frequencies within these limits. Each time, the line (rssi peak) and the cursor were accurately positioned.

So, for the time being, it will be difficult for me to explore this question further. 

Armel F4HWN.

DL2MF, 2024-11-25:
Hi Armel,

yes, it's really a nightmare to reproduce this bug, today I did test again during some sondeflights. See the results, it looks really very strange on 405.300 MHz and then later again at 403.310 MHz everything was fine from scratch again.

[image]
Sample 1: This was my todays start screen - the marker seems to be at the right position, but the maximum is at the right border.

[image]
Sample 2: Changing the stepping/decimation produces better spectrum but marker lags at faulty pos.

[image]
Sample 3: My nearest launch site at 405.700 MHz also started, but marker and freq doesnt fit.

[image]
Sample 4: Playing again with steps brings back the correct spectrum / marker positions.

[image]
Sample 5: Also with 10k Sondeband steps now the spectrum and marker are fine.

[image]
Sample 6: One hour later when I tuned to 402.370 - 403.650 the next sonde at 403.310 MHz is also correct displayed.

I have enough radios available to do some deeper investigation of this effects. I'll try to find a simple step-by-step description, how to get the radio so confused in the spectrum screen ;-)  

Best regards,
Mike, DL2MD

armel, 2024-11-25:
Hello Mike,

Yes, it's very strange but interesting :) If you can find a way to reproduce the problem every time, it would be easier for me to analyze it. 

Regards,

Armel F4HWN.

GonzoG, 2024-12-01:
I've got this too. It's the bars only. Marker arrow is in correct position.
It looks like the bars are wrong width (too wide) for 16-64x. For 128x it's ok. Bars are 1pix.
I've set spectrum to 32x and it looks like it shows only 8 bars. If you look at photos posted DL2MF, at 64x bars should be 2pix wide, and they are 4 pix wide.

Manka76HUN, 2024-12-03:
[image]
After version 3.6 the column width is not much larger. 2 columns appear with the same setting.

armel, 2024-12-03:
@Manka76HUN And ? If the RSSI is the same... where is the problem ?

Manka76HUN, 2024-12-03:
> @Manka76HUN And ? If the RSSI is the same... where is the problem ?

My only issue is that multiple columns are more comprehensible to me than just two.😀

jflinchbaugh, 2024-12-25:
This might be an aspect of this problem:
1. Set a scan range of 144MHz to 148MHz using VFO A and VFO B and long-pressing 5.
2. Fn-5 to launch specrtrum analyzer.
3. Set it to 5.00k jumps, and the samples stay at 800x. 
The spectrum graph renders narrow on the left of the screen instead of the full width, and you can see the reception peaks in there.
The pointer at the bottom below the spectrum graph seems to span the whole screen when showing an incoming signal.

I've observed this problem in 3.7, 3.8, and 3.9.

[image]

armel, 2024-12-25:
@jflinchbaugh No, I explained why you observed this in [one](https://www.youtube.com/watch?v=y8oe1odsU88) of my videos (in french sorry). In addition, choosing 5kHz as a step doesn't really seem justified. And don't ask too much of this little bandscope, which fits into just a few KB of memory 😌

jflinchbaugh, 2024-12-27:
I choose 5kHz steps for scanning, so I don't skip over an active frequency. Am I thinking about that parameter incorrectly? 

Other firmwares, like egzumer and aubsuk, scale the graph fine when I did this range. Are those different implementations of the graph that take up more memory? 

Thanks.

armel, 2024-12-27:
Ok @jflinchbaugh :)

So, feel free to use Egzumer or Aubsuk (I don't know this one) if it suits you :) We're lucky enough to be able to choose between a lot of firmware, which is great. However, you might encounter cases where it won't work. I've made a few changes to improve the Egzumer one. 

Also, using a 5 kHz step doesn’t make much sense. Check the recommended step on VHF for your region, as it depends on your country. I'm using 12.5 kHz in France. 

Armel.

jflinchbaugh, 2024-12-31:
> So, feel free to use Egzumer or Aubsuk (I don't know this one) if it suits you :) We're lucky enough to be able to choose between a lot of firmware, which is great. However, you might encounter cases where it won't work. I've made a few changes to improve the Egzumer one.

Yes, I've tried lots of firmwares, and so far this is the overall best I've found for now. I appreciate seeing regular updates.

armel, 2025-01-07:
@franck-x By the way, your piece of code doesn't solve the problem... Nice try. But I encourage you to come back when you have real solutions, rather than a hasty judgment of my work. I have always been open to contributions when they genuinely improve things. However, I also reserve the right to reject them if I believe they add no value. 

So this issue is still **open**, don't hesitate to contribute. 

Thank you again.

Armel, F4HWN

jflinchbaugh, 2025-01-07:
I've not taken time yet to understand why the F4WHN-specific DrawSpectrum is different, but I kept all the other stuff and reverted back to the other DrawSpectrum implementation that's there:
```
-#ifdef ENABLE_FEAT_F4HWN_SPECTRUM
+#ifdef ENABLE_FEAT_F4HWN_SPECTRUMx
     static void DrawSpectrum()
     {
```
I rebuilt and deployed it to my radio, and for my case  (144MHz-148MHz, step 5khz, 800x), I see the spectrum scales to the  width of the screen just fine. The spectrum and marker stay lined up with larger step values too.
What's the benefit of the F4HWN-speciifc spectrum vs the old? Is there a middle ground to be had? 

[image]
[image]

jflinchbaugh, 2025-01-07:
Oh, I just tried the really narrow test 405-405.5, and this old implementation I'm trying now shows the graph narrow. Hmm. It's like it needs one implementation for extra-wide ranges, and one for extra-narrow.

armel, 2025-01-07:
Hello @jflinchbaugh,

> Oh, I just tried the really narrow test 405-405.5, and this old implementation I'm trying now shows the graph narrow. Hmm. It's like it needs one implementation for extra-wide ranges, and one for extra-narrow.

Yes, the change you're talking about came from [this PR ] (https://github.com/armel/uv-k5-firmware-custom/pull/295) to solve issue #285. It solved it, but it doesn't solve the current issue #322. 

Armel, F4HWN.

jflinchbaugh, 2025-01-08:
https://github.com/armel/uv-k5-firmware-custom/pull/352

armel, 2025-01-08:
Hello @jflinchbaugh,

> I think I have a fix that accommodates narrow ranges and wide ranges. #352

Okay, I'll review, check and test tonight. Thanks for your contribution.

Armel, F4HWN.

armel, 2025-01-08:
Hello @DL2MF, 

Please, could you test it and let us know if it works for you too ?
Here is a compiled firmware [f4hwn.bandscope.packed.bin.zip](https://github.com/user-attachments/files/18347608/f4hwn.bandscope.packed.bin.zip)
Please note that, due to GitHub's security limitations, this is a **zipped** binary. You'll **need to unzip it before using it**.

Thank you,

Armel, F4HWN

armel, 2025-01-08:
Merge on [feature_update_v4](https://github.com/armel/uv-k5-firmware-custom/tree/feature_update_v4). 
Will be fixed in next release. 
Good job @jflinchbaugh and thank a lot for this contribution.

Armel, F4HWN

commit bd0911f7 (2025-01-08): Merge pull request #352 from jflinchbaugh/main

fix #322: scale bandscope for any number of samples
commit d0da9ac7 (2025-01-08): fix #322: scale bandscope for any number of samples

retain stretch of bars for short ranges, but after 128,
scale 1:1 to always fill the width of the screen.
closed 2025-01-08 by armel