armel/uv-k5-firmware-custom #326: `POWER HIGH` and `REMOVE OFFSET` actions should be reset when changing channel

emanuelegissi, 2024-12-04:
Hi Armel,
I stumbled upon what I consider a risky behaviour of the `POWER HIGH` and `REMOVE OFFSET` actions.

An example:
- Let's say that the rescue operator is working on channel M123, and that channel has `Duplex` activated by default.
- Then a direct communication is needed. The operator pushes the `F2` button for the `REMOVE OFFSET` action, and obtains a direct communication.
- Then the operator switches to channel M124, also that channel has `Duplex` activated by default.
 - The current 3.8 firmware does not reset the `REMOVE OFFSET` action, so `Duplex` is removed on channel M124 too.

The same behaviour happens with `POWER HIGH`.

I suggest resetting both `POWER HIGH` and `REMOVE OFFSET` actions, when changing channel.

The default channel configuration should always have priority: the `POWER HIGH` and `REMOVE OFFSET` actions should only be temporary exceptions, knowingly chosen by the user on that currently selected channel.

And even if the user goes back to the original M123 channel, the default M123 channel configuration should have priority, no need to save the previously modified state.

What do you think about this?
Thank you

armel, 2024-12-05:
Hello @emanuelegissi 

I agree. I understand and share your opinion on the risks. No problem. 
It's [fixed](https://github.com/armel/uv-k5-firmware-custom/commit/99de42daebb6d58f9e506ee361eaaafc9d9e24a5).  I'll let you recompile and test.

Keep me updated ;)

Armel F4HWN.

emanuelegissi, 2024-12-06:
Uploaded the fixed hardware. I confirm it works well. IMHO this issue can be closed. Thank you

armel, 2024-12-06:
Hello @emanuelegissi, ok thank you.

closed 2024-12-06 by armel