armel/uv-k5-firmware-custom #327: Analyseur de spectre affiché = pas de mise en veille ?

F6GDN, 2024-12-05:
Bonjour Armel,

Encore et toujours merci pour le travail formidable sur le firmware.
Que de chemin parcouru depuis le début.
Sauf erreur ou omission de ma part, il apparaît sur mon UV-K5 que lorsque je laisse l'affichage en mode analyseur de spectre, la mise en veille ne se fait pas.
Pour tous les autres modes d'affichage, cela fonctionne.
Au plaisir de te lire à ce sujet.
Passe une excellente journée !
73 !
Eddie
F6GDN

armel, 2024-12-05:
Bonjour Eddie,

Merci ;)

Oui, c'est effectivement le cas, tu as raison et c'est normal.

L'analyseur de spectre est une application à part, embarquée dans le firmware. Il dispose de sa propre gestion du clavier et des raccourcis, pas de powersave et aucun mécanisme de mise en veille, etc. L'analyseur de spectre ne prend pas en compte ce type de paramétrage et ne passe plus dans les boucles de temporisation qui gèrent certains comportement du poste (comme la mise en veille). Bref, il est autonome. 

Je te donne un autre exemple:
- règle BLTime (menu 38) sur 10s (par exemple)
- règle BLMin (menu 39) sur 1 (par exemple)
- règle BLMax (menu 40) sur 6 (par exemple)

Hors analyseur de spectre, avec ces paramétrages, tu verras la luminosité de l'écran baisser au bout de 10s sans action sur le poste (passant de 6 Max à 1 Min). Mais si tu bascules dans l'analyseur de spectre, l'écran restera allumé, même au delà de 10s. 

Voilà, tu sais tout. 

Armel F4HWN

closed 2024-12-05 by F6GDN

armel, 2024-12-05:
Avec plaisir Eddie et merci encore ;) 

Armel F4HWN.