armel/uv-k5-firmware-custom #328: Radio wave transmission prohibition is not working.

hirokane8086, 2024-12-05:
Radio wave transmission prohibition is not working.

armel, 2024-12-05:
First of all, good morning—it's free to say it.

Secondly, could you be a bit more specific? Coming here and saying that a certain feature doesn't work is pointless. Give us some details.

In the end, we'll quickly see that you either didn’t read or didn’t understand the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-flock-and-txlock-menus). I'm willing to bet on it...

Armel F4HWN.

armel, 2024-12-06:
Hello @SHODAN556,

Thank you for your message. 

Selecting DISABLE ALL from the F Lock menu is a good idea, especially if you don't have a callsign. Good job ! But if you've already tried another firmware (like Egzumer), it's not enough. Now you need to check the TXLock parameter which is specific **to each memory channel**. 

If TXLock is ON, transmission will not be possible. If TXLock is OFF, transmission will be possible. You can control TXLock parameter from the transceiver, but more effectively from Chirp (using the [dedicated driver](https://github.com/armel/uv-k5-chirp-driver)). Note that, it's possible to make a [multiline selection from Chirp](https://youtu.be/ZdN9FlEwlOg?t=330) and modify the TXLock parameter on several channels with just a few clicks.

Armel F4HWN.

SHODAN556, 2024-12-06:
Hello @armel 
IMHO, as a one week Quansheng owner who just wanted to check is something interesting in the air in my city to listen to (and realized city of mine is dead-silent...)  "DISABLE ALL" should disable every TX on this device. This particular setting should be idiot(me)-proof.
Maybe it would be a good idea to reverse the process? When user sets F LOCK to "DISABLE ALL" (which means he/she REALLY do NOT want to mess anything in the air, wants just to satisfy his/her curiosity in safe way) there will be TXLock setting "ON" on every frequency? User would have to consciously, once, set TXLock to "OFF" in certain range.

armel, 2024-12-06:
I understand @SHODAN556. But it's so easy to control and set TXLock from Chirp and multiline selection. Plus, it's a setting you don't make every day, but once when you first initialize your transceiver. It also saves precious flash memory, which, as you know, is very limited. Forcing TXLock to ON when DISABLE ALL is selected would still take up memory space.

hirokane8086, 2024-12-07:
I think I should add a setting to make it a receiver. UV-K5 will create a new radio operator. There are many beginners.
All you have to do is make it selectable in the Makefile. I think there are too many choices. Is there a lot of movement in continental countries? I live in an island country and don't know that. In the case of JAPAN, it is the same as CE. However, many people use UV-K5 as a receiver. This is because spurious regulations are strict in Japan. When emitting radio waves, use YAESU or ICOM.

armel, 2024-12-08:
Ok guys, in the next version 3.9, if we select F Lock DISABLE ALL and validate this choice, it will force the TxLock field to be reset to ON for **all** channels. Of course, after that, you're free to set TxLock to OFF on the few channels on which you want to be able to transmit. 

Armel.

hirokane8086, 2024-12-10:
DISABLE ALL has no meaning if you can select it and send it. There are also options to use other firmware.

hirokane8086, 2024-12-10:
Are you aware of any bugs that allow sending even if DISABLE ALL is selected? From my research it seems like it can transmit on the default frequency range. It's not written in the manual. Your opinion is that it consumes a lot of memory, but why did you increase the CA? You should select it in MAKEFAILE. I'm not saying to add JAPAN, I'm saying to make sure the setting that prohibits all transmissions for people who don't have a license works properly.

armel, 2024-12-10:
@hirokane8086 Sorry, I don't understand your comment. 

Again **If you select DISABLE ALL and if your memory channels / VFO have TXLock set to ON**, it is not possible to switch to transmit. If there is a bug here, tell me. 

About Japan, I could add a new F Lock zone as I made for Canada. But nobody asked me.

Seriously, if you're afraid to transmit with this transceiver, why are you using it? It's a bit like asking a gun manufacturer to make sure you can't shoot it.

Armel F4HWN.

SHODAN556, 2024-12-10:
@armel 
If it is not a bug for you, I think we should call it "wrong way to apply the feature".
In EGZUMER 0.22 "DISABLE ALL" meant "DESOLDERING TX MODULE RIGHT NOW!"
TBH F Lock setting in this firmware is worthless. Setting "DISABLE ALL" is doing literally nothing. I can still transmit until I will set TxLock to "ON". It gives false sense of security, it can still put me into jail, because it may happen I will accidentally transmit where I would like just to listen and where I must not to transmit, cause I don't have a callsign (In Poland it is legal to listen everything until you don't break any encryption or any other protection like TETRA etc., transmit requires licenses and callsigns.)
As you said it is sent once per range - but I have to do it on EVERY range, MANUALLY.
As you said it can be done in CHIRP - I'm not using CHIRP AT ALL, because I prefer scanning the range manually, by hand, on radio.

>About Japan, I could add a new F Lock zone as I made for Canada. But nobody asked me.

We just need one settting which will work always, everywhere. We need "SHUT THE FUCK UP!" checkbox.

>Seriously, if you're afraid to transmit with this transceiver, why are you using it? It's a bit like asking a gun manufacturer to make sure you can't shoot it.

What if me/he/whoever wants just to RECEIVE, not TRANSMIT? Quansheng is dirt cheap and it can be used just for scanning range/listening to interesting chatter on shortwave or PMR. I bought that radio just to check is there something interesting to listen in my city. I just realized I could buy RTL-SDR for that and use it without a fear of going to jail, but Quansheng went 2-3-4 times cheaper than that.
Your comparison radio to a gun is wrong. You can break the law or NOT break the law with these things. You can use a radio just to listen what is going on in the city or you can (purposely or not) disrupt work of many city services as police or, as it happened in Poland, stop the nearest trains just by TXing RADIO-STOP signal. With a gun you can practice your aim on shooting range, but also you can make a bloodbath on the streets. 
Your call and your responsibility.
Just give us a one, reliable, easy to set, easy to enable, simple and botherless way to shut every TX down.

armel, 2024-12-10:
Seriously, once again, learn how to use this transceiver and firmware:

1 - F Lock -> DISABLE ALL 
2 - TXLock -> ON (on VFO and all your channels)

If you can't, use another firmware. 

It's clear that you don't know the background to this issue. You come here without knowing the context and without any desire to understand. The TXLock parameter was the only solution allowing exceptions based on individual use cases, various specific scenarios and country-specific regulations (PMR, FRS, GMRS, MURS, FFVL, and others). Previously, the only option for users was to UNLOCK EVERYTHING. Do you really think that was better? Then wake up and stop complaining.

As I said in the README:

THIS FIRMWARE HAS NO REAL BRAIN. PLEASE USE YOUR OWN. Use this firmware at your own risk (entirely). There is absolutely no guarantee that it will work in any way shape or form on your radio(s), it may even brick your radio(s), in which case, you'd need to buy another radio. Anyway, have fun.

I'm closing this issue that leads nowhere. That's enough. 

Armel F4HWN.

closed 2024-12-10 by armel