armel/uv-k5-firmware-custom #332: CCIR identification and selective call

emanuelegissi, 2024-12-11:
Our official handheld radios have self identification and selective call using the CCIR protocol (5 or 6 tones):
https://en.wikipedia.org/wiki/Selcall

The tone duration for identification is 50 milliseconds, the tones are transmitted sequentially at the beginning of each tx. The tone duration for selective calling is 100 milliseconds.

I have seen that the IJV firmware has implemented this feature (search for "CCIR"):
https://www.universirius.com/en_gb/preppers/quansheng-uv-k5-manuale-del-firmware-ijv_v3/

and explanation video:
https://youtu.be/lm6FbR7DWvg?si=o2ljhuVM4s_cSVBh

 I wrote to them kindly asking to share the relevant code. I will keep you posted on that.

armel, 2024-12-11:
Hello @emanuelegissi,

Okay, I don't have any experience of the CCIR protocol. I've never studied it. But I understand the use case. 

Maybe IJV can help here, although it will certainly be complicated to adapt from one firmware to another. 

Keep me informed. 

Armel F4HWN.

MrHoerby, 2024-12-12:
Just for understanding.
[Uploading Grundig SCM 1 Selektiv Ruf BDA Schaltplan.pdf…]()

emanuelegissi, 2024-12-13:
I received a kind answer from IJV, but no mention on sharing any code, so **no help available**.

In the meantime:
- I took a look at the existing dtmf functionality and how it is implemented in egzumer (in armel version it is disabled).
- I read the IJV manual at https://www.universirius.com/en_gb/preppers/quansheng-uv-k5-manuale-del-firmware-ijv_v3/

I am not sure I understood everything, but it seems to me that all that is needed is already there.

I have the detailed specification on how our system works, and it seems like a rather standard CCIR implementation.

For your curiosity, our radios work like this:

- the **identification code** is sent at the beginning and at the end of the transmission, like this:
250ms silence + 5x 50ms tones + 60ms silence

- the **selective call code** is sent like this:
1500ms silence + 5x 100ms tones + the identification code

- the positive **answer** to the selective call is:
800ms silence + 3000ms D tone (991Hz) + the identification code

Here is the digit tones table:
0: 1981 Hz
1: 1124 Hz
2: 1197 Hz
3: 1275 Hz
4: 1358 Hz
5: 1446 Hz
6: 1540 Hz
7: 1640 Hz
8: 1747 Hz
9: 1860 Hz
A: 2400 Hz
B: 930 Hz
C: 2247 Hz
D: 991 Hz
E: 2110 Hz
F: 1055 Hz

As the specific use case is rather limited, maybe I could try to fork the code and implement it, replacing the dtmf functionality.

But maybe it would be more useful to the wider community if we had an implementation that was customizable at compilation time. For example a json file with the desired digit-tone frequency table, and another one with identification, selective, and answer code sequences. No firmware that I know as anything like this.

Just my 2 cents. What do you think?

armel, 2024-12-13:
Hello @emanuelegissi,

Ok about IJV. That's his choice, even if I think he would have done better to open source his code. 

Returning to the main topic, I’ve indeed disabled the DTMF Calling feature a long time ago. It doesn’t have much relevance in the world of amateur radio communications. That said, I can see how it could be useful in the context of how you’re using this little K5.

It is probably possible to adapt the code for the DTMF Calling feature to add CCIR support. But be aware:

- Current DTMF Calling implementation requires configuration via Chirp (e.g., setting the ANI ID, DTMF Contacts, etc.). You would likely need to update the Chirp driver in addition to modifying the firmware,
- Display issues may occur, such as overlapping text. I’ve made numerous changes to how information is displayed depending on the RxMode. Be aware,
- Forget the JSON-based approach, as it doesn't seem appropriate here.

Here’s what I think would be the best approach:

- Fork the current firmware,
- Add a new directive to the `Makefile` (e.g., `ENABLE_CCIV_CALLING`),
- Modify the `compile-with-docker` script to include something like the following in the `rescueops()` function:

```
rescueops() {
    echo "RescueOps compilation..."
    docker run --rm -v "${PWD}/compiled-firmware:/app/compiled-firmware" $IMAGE_NAME /bin/bash -c "cd /app && make -s \
        ENABLE_SPECTRUM=0 \
        ENABLE_FMRADIO=0 \
        ENABLE_AIRCOPY=1 \
        ENABLE_NOAA=1 \
        ENABLE_CCIV_CALLING=1 \
        ENABLE_FEAT_F4HWN_RESCUE_OPS=1 \
        EDITION_STRING=RescueOps \
        TARGET=f4hwn.rescueops \
        && cp f4hwn.rescueops* compiled-firmware/"
}
```
- Update the firmware code, adding conditional sections like:

```
#ifdef ENABLE_CCIV_CALLING
// CCIV Calling feature code here
#endif
```

Keep in mind that this is probably a big job.

But if it works well and the code isn't too intrusive, it will probably be possible to merge it into the current firmware code via a PR (Pull Request). 

Regards,

Armel F4HWN.

emanuelegissi, 2024-12-15:
Okay Armel, thank you for your guidance. I will let you know.

MrHoerby, 2025-01-04:
[I found this on this page: ](https://recon.ddns.net/)
https://recon.ddns.net/
[image]

emanuelegissi, 2025-01-07:
Thank you, very interesting! Do you know if the code is open source and available?

MrHoerby, 2025-01-07:
I think it comes here, I've already searched a bit but haven't found anything yet. With a bit of luck you can find the code in one of the
source code files: https://github.com/RE3CON?tab=repositories

markusb, 2025-01-17:
I have a similar issue but for ZVEI2 calling. This is used in Switzerland for the REGA emergency call network. For this only two defined 5-tome ZVEI sequences are reuired. On for a test call and une for a real call. So no configurable tone sequences are needed.

But the question is more how to integrate the function in a easy way. One could even imagine to hardcode it to the corresponding frequency or a channel name, to make it simple.

ttimpe, 2025-08-10:
It would be amazing if I could program a button to transmit a specific ZVEI 5-tone sequence and unsquelch on another one (or multiple, as group calling is sometimes needed). This is/was a standard feature of many portable radios such as the ones from AEG/Telefunken or Bosch.