armel/uv-k5-firmware-custom #333: exit/D buton TX

cziss, 2024-12-12:
Hi, I'm experiencing strange behaviour, which is not (I believe) documented on wiki. Pushing exit/D buton transmit signal, some kind of beep. I think that's very unconvinced, because it's button with huge usage an it should broadcast anything.

cziss, 2024-12-12:
Well, it seems that pushing M and UP and DOWN also trasmit something. Also when you are browsing menu.

cziss, 2024-12-12:
https://github.com/user-attachments/assets/a556d1f0-1e01-4666-8a7c-b480615e0288

I've made video of that.

cziss, 2024-12-12:
https://github.com/user-attachments/assets/feabba01-98e8-40da-8ed5-31d5ee16dd05

I've made video of that.

cziss, 2024-12-12:
https://github.com/user-attachments/assets/0bc79223-c836-47f9-a01a-b467a12830ba

I've made video of that.

armel, 2024-12-12:
Hello,

That's simply because you've activated vox (menu 53). So a single keystroke can be picked up by the nearby microphone...

Armel F4HWN

closed 2024-12-12 by cziss