armel/uv-k5-firmware-custom #337: Bug Squelch Tail Eliminator version SA818

F4HWD, 2024-12-21:
Bonjour Armel,

J'utilise le firmware `f4hwn.bandscope.packed.v3.8.sa818.bin` sur mon UV-K6. Le Squelch Tail Eliminator fonctionne bien, contrairement au firmware standard (nous en avons discuté par mail).
Sauf que, j'ai constaté un problème aujourd'hui : lorsque je démarre mon talkie, il revient sur le dernier canal utilisé, la plupart du temps celui de mon hotspot. Jusque là tout va bien. Par contre, dans ce cas de figure, j'entend la queue de squelch lorsque le hotspot cesse son émission. La seule solution pour régler le problème est de changer de canal, puis de revenir sur celui du hotspot. Et là, plus de soucis, je n'entends plus rien quand mon hotspot repasse en Rx.

As-tu une solution à proposer ? Merci d'avance.

73's de F4HWD

armel, 2024-12-21:
Bonsoir Maxime,

Ma réponse en vidéo, ce sera plus simple.

https://github.com/user-attachments/assets/c91b4e80-898f-4e63-8363-9d5381d3a355

Je n'arrive donc pas à reproduire ce que tu décris, désolé. 

73'

Armel F4HWN.

F4HWD, 2024-12-21:
Merci pour ta réponse très rapide.

J'ai trouvé pourquoi tu n'arrives pas à reproduire le bug : tu es en Dual Watch. J'arrive au même résultat que toi.

Par contre, si tu passes le poste en Main Only dès le démarrage de celui-ci, le bug est bien présent.

Je te laisse tester et ferais une vidéo si besoin. En tout cas bon courage, ce bug a l'air bien vicieux.

73's de F4HWD

armel, 2024-12-22:
Pas mieux :)

https://github.com/user-attachments/assets/b91b8dc7-22a0-43a0-b673-b433b9ff5792

73'

Armel F4HWN

F4HWD, 2024-12-22:
J'ai trouvé d'où vient le problème : si tu as un code CTCSS en réception sur le Quansheng, le bug se produit. Si tu ne l'as pas eu, c'est parce que tu n'as mis aucun code en réception.

Essaye d'activer l'émission du CTCSS sur ton hotspot et active ensuite la réception du code sur ton UV-K5, tu auras le même problème que moi (voir la vidéo).

https://github.com/user-attachments/assets/9e5a3d08-c8d3-426b-89eb-576920177ce7

73’s de F4HWD

armel, 2024-12-22:
Ok, je ne suis donc pas étonné avec un CTCSS en réception. Comme quoi, tous les détails compte quand il est question de résoudre un problème.

Donc c'est trivial, dans radio.c, juste après la ligne [775](https://github.com/armel/uv-k5-firmware-custom/blob/feature_update_v3/radio.c#L775), ajoute : 

```
BK4819_SetTailDetection(SQL_TONE); // Default 550 = QS's 55Hz tone method
```

C'est aussi simple que cela.

Armel F4HWN.

F4HWD, 2024-12-22:
En effet, la solution était simple. Je ne peux hélas pas la tester, je n'arrive pas à reproduire le firmware sa818 avec Docker, il reste quoi qu'il arrive bloqué avec le STE d'origine à 55 Hz. J'ai rajouté une ligne avec SQL_TONE=600 \ dans compile-with-docker.sh, sans succès.

C'est pas grave, je vais réinstaller la dernière version stable et attendrais la version 3.9 pour avoir le correctif, que tu peux d'ores et déjà valider, après lecture du code, ça devrait fonctionner je pense.

Merci en tout cas, je clôture l'issue et la rouvrirais si le problème persiste avec la 3.9

73's de F4HWD

closed 2024-12-22 by F4HWD

armel, 2024-12-22:
C'est normal vu la _portée_ de `SQL_TONE`. Tu ne peux que le changer dans le `Makefile`.
Ca, je vais probablement le modifier.

Par contre, vu l'intérêt proche du **zéro +∞** autour de la version SA818, je doute aller plus loin. Il faut bien se rendre à l'évidence: tout le monde s'en cogne du SA818 et autre Shari :) En parallèle de mes expérimentations perso, tu es le seul à t'en préoccuper. 

Je n'ai eu _aucun_ retour concernant la version SA818 (autre que les tiens). Par contre, pour en causer des heures sur certains réseaux _hautement_ qualifié (HI), ca, il y a du monde. Tout ca alors que ca fait des mois que je diffuse une version patchée.

C'est dire l'étendu du désert autour de cette question. L'explication vient probablement du fait que 99% des OMs (si on peut encore les appeler comme cela, vu le niveau d'expertise...) sont _incapables_ d'activer le `+DMOSETTAIL:0 tail: open`. 

73'

Armel F4HWN

F4HWD, 2024-12-23:
Je comprends très bien ce que tu veux dire ;) Après c'est dommage car cette fonction apporte un réel confort d'écoute. Je connais pour ma part des OM qui l'utilisent, mais nous déplorons tous la même chose : que l'option ne soit pas intégrée sur tous les logiciels de configuration du SA818. Elle est intégrée sur ASL3 par exemple, mais pas dans la distribution Spotnik, en tout cas pas de façon simple.

Pour en revenir au bug, je confirme que le patch publié hier résout le problème de compilation avec Docker. Par contre, j'ai du rajouter la ligne que tu m'as indiquée plus haut dans `radio.c`, tu as oublié de patcher le fichier :) Du coup l'ensemble fonctionne correctement maintenant, plus de queue de squelch.

Pour pérenniser ce système, je me permets de te faire 2 propositions : 

- Soit : dans `compile-with-docker.sh`, rajouter `SQL_TONE=550 \` dans toutes les versions compilées, ou bien au moins la version bandscope. Ceux qui veulent garder le STE à 55 Hz (standard) n'auront rien à toucher et cela ne t'impactera pas non plus dans tes projets. Les 1% de "vrais" OM (humour à prendre au 1000e degré) iront modifier cette valeur à la main en mettant 600 et recompileront eux-mêmes.

- Soit : toujours dans ce même fichier, rajouter un firmware bandscope_sa818 qui aura directement la bonne valeur pour le SA818. Un peu comme le firmware que tu sors actuellement, mais ce sera automatisé cette fois.

Te connaissant, je pense que tu préféreras plutôt la première solution ;)

73's de F4HWD

armel, 2024-12-23:
Bonsoir Maxime, 

Comme dit hier, je n'ai pas prévu d'intégrer le patch pour le cas du CTCSS en RX/TX. Je t'ai juste indiqué comment faire, _mais je n'irais pas plus loin_. Ce n'est donc pas un oubli, c'est volontaire, sans compter que je manque de place mémoire actuellement. 

Quant à tes 2 propositions, c'est volontaire aussi. Encore une fois, le cas particulier du SA818 n'intéresse personne. Je préfère éviter d'y faire référence le plus possible. Mais j'aviserais peut-être plus tard.

Dans les faits, quand je package les versions SA818, je me contente de changer la valeur du `SQL_TONE` dans le `Makefile,` je lance un `./compile-with-docker.sh all` et c'est tout. Je le fais qu'une fois, à chaque nouvelle version de production (donc genre 1 ou 2 fois par mois...). 

73'

Armel F4HWN.