armel/uv-k5-firmware-custom #338: [Feature] Wiki in other lanquage

ErikS-web, 2024-12-23:
Hello,

nice work you've done on firmware development and this Wiki.
I read you point to specific translated manuals, but ONLINE the Wiki is only in English (+French) lanquage

With this line you can provide **Online Wiki in other lanquage** with Online translation with in Github.

https://github-com.translate.goog/armel/uv-k5-firmware-custom/wiki?_x_tr_sl=en&_x_tr_tl=nl

Suggest to add some lines on Home of Wiki 
Here is the format (EDIT this issie to see it)

## Translations

The Wiki can be translated into other languages using Google translate. Four languages are predefined below, but if you open one of them, you will be able to select other languages at the top of the translated page.

* [POLSKIE TŁUMACZENIE / Polish translation](https://github-com.translate.goog/armel/uv-k5-firmware-custom/wiki?_x_tr_sl=en&_x_tr_tl=pl&_x_tr_hl=pl)
* [Nederlandse vertaling / Dutch translation](https://github-com.translate.goog/armel/uv-k5-firmware-custom/wiki?_x_tr_sl=en&_x_tr_tl=nl)
* [Traduzione italiana / Italian translation](https://github-com.translate.goog/armel/uv-k5-firmware-custom/wiki?_x_tr_sl=en&_x_tr_tl=it)
* [Traduction française / French translation]( https://github-com.translate.goog/armel/uv-k5-firmware-custom/wiki?_x_tr_sl=en&_x_tr_tl=fr)

If you need help, just give a reply.

Greetings Erik

armel, 2024-12-24:
Hello Erik,

Good idea.
Done !

Armel F4HWN.

ErikS-web, 2024-12-24:
Top.

closed 2024-12-24 by ErikS-web