armel/uv-k5-firmware-custom #339: M24M02 EEPROM management (11151 channels)

UserAZERTY0, 2024-12-23:
Hi,
Would it be possible to support bigger size EEPROM so that additional channels can be managed (+200) ?
Of course this would require a small hardware modification but as it is a rather simple one (compared to Si4732 for instance), it would be still very usedfull for most of us. 
Hardware modification explained : https://www.youtube.com/watch?v=Sg41OMVMMUM
Thank you.

armel, 2024-12-23:
Hi,

No, this is not planned.

It would require a complete rewrite of the settings save process, major changes to the firmware code, and adjustments to the Chirp driver. In the end, we'd probably end up maintaining two versions of the firmware and two versions of the Chirp driver. So, no thanks.

It would be a huge amount of work to satisfy only a handful of users, especially since 99% of K5 owners have only 8KB of EEPROM.

But feel free to do it yourself if you want, the code is open source.

Armel F4HWN.

ErikS-web, 2024-12-24:
If you would like to have 'unlimited' (group-)channels, I suggest to run UVK5 from PC.

For more info see this [Q-Dock-project](https://github.com/nicsure/QuanshengDock/wiki) and enjoy the Wiki i've made for it.