armel/uv-k5-firmware-custom #342: Ver. 3.9 (RescueOps). NOAA channels doesn't have the right bandwitdh

simonforget, 2024-12-25:
Hi,

I just updated to version 3.9 (RescueOps) and noticed the NOAA channels bandwidth is set by default to 12.5kHz, while it should be set by default to 25kHz.

Keep up your excellent work!

Simon Forget, VA2ESF

P.S.: I'm using English only to make sure that more people understands the problem that I'm reporting. Autrement, je suis plus à l'aise en français. :-)

armel, 2024-12-25:
Hi,

It doesn't matter. If you use the UP or DOWN keys to change frequencies, you are cycling between the different NOAA [frequencies](https://github.com/armel/uv-k5-firmware-custom/blob/main/frequencies.c#L48-L62). 

Armel F4HWN.

closed 2024-12-27 by armel

simonforget, 2025-01-26:
Bonjour Armel,

J'espère que vous recevrez ce courriel.

J'ai regardé avec attention vos deux dernières vidéos concernant le
Quansheng TK-11.

En ce qui concerne la limitation sur le nombre de canaux FM commerciaux
(broadcast) qui est limité à 20 sur les Tk-11, c'est en fait la limite qui
existe en Amérique du Nord. Notre bande FM commerciale se situe entre 88 et
108 MHz.

Continuez votre excellent travail!

73,

Simon Forget, VA2ESF
***@***.*** ***@***.***>
Keybase: sforget
Signal: sforget.07
Discord: sforget
M: +1 (514) 742-4400

armel, 2025-01-26:
Bonsoir Simon,

Hummm, je doute que les chinois se préoccupent de limitations en vigueur en Amérique du Nord :)
En France, rien que sur Paris, il existe plus de 60 stations Broadcast FM.
Mais attendons de voir ce que proposera ce TK11, quand il sera disponible.

73'

Armel, F4HWN.

simonforget, 2025-01-27:
Bonjour Armel,

Sincèrement, je ne serais pas surpris que ces vingt stations de radios FM
possibles dans un marché nord-américain soient la seule raison pour ces
vingt mémoires.

Le plan de fréquence des stations commerciales FM en Amérique du Nord est
entre 88 et 108 MHz. Chaque station est espacée de 200 kHz, étalonnées sur
les centaines de kHz impaires (ex. 95,1, 96,9, 107,3...). En tenant compte
de la réassignation, on arrive à vingt stations par marché majeur.

Selon mes souvenirs, les fréquences entre 76 et 87 MHz étaient assignées à
la télévision terrestre analogique avant la transformation vers le
numérique. Ces fréquences n'ont pas été réassignées en ce moment; une
rumeur persistante court sur l'assignation d'une partie de ce spectre aux
radioamateurs (4m).

73,

Simon Forget, VA2ESF
***@***.*** ***@***.***>
Keybase: sforget
Signal: sforget.07
Discord: sforget
M: +1 (514) 742-4400