armel/uv-k5-firmware-custom #344: Squelsh au niveau du BandScope

xfailed, 2024-12-26:
Bonjour je me permet de mettre un petit message ici concernant le firmware avec le bandscope. J'ai remarqué,  c'est pas d'une grande importance mais un petit peut irritant. quand on est en mode normal et que cesse une émission sur une fréquence quelconque le squelsh coupe de suite la fréquence ce qui fait que le bruit est de suite coupé . Par contre quand on est en bandscope quand cesse une emmission le squelsh ne coupe qu'au bout d'une seconde ou 2 ce qui fait qu'on entend le bruit FM pendant une ou 2 seconde et ça m'irrite. Alors c'est peut être ça ne dérange que moi mais est il possible de réduire le temps de réponse du squelch au niveau du Bandscope ?
En tout cas bravo pour ce Firmware que je suis avec attention.

armel, 2024-12-27:
Bonsoir,

Ok, je vois, mais j'ignore totalement comment et si je serais en mesure de fixer ce comportement. Du reste, ca ne me dérange pas du tout. Les queues de Squelch sont compliquées a gérer et filtrer et font partie (des charmes) de la radio.

73'

Armel F4HWN.

xfailed, 2024-12-29:
Bonjour, ... Bon ce n'est pas grave c'est juste que quand il y a un QSO avec une radio assez faible j'augmente le son et le souffle après le relâchement du PTT de l'émetteur est assez fort et ce Pshiiiit !! est aussi entendu par les personne qui sont relativement proche de moi et ça les agace, du coup je suis obligé de baisser le volume et je n'entend pas les radio faible. Comme je dis ça ne concerne que moi alors si c'est trop compliqué, je vais essayer autre chose. Notamment ne pas utiliser le bandscope mais le scan.
Merci quand même.

closed 2024-12-29 by armel