armel/uv-k5-firmware-custom #346: exclusion de mémoire temporarire

stephane-rib, 2024-12-27:
Bonjour Armel et tout d'abord merci encore pour le travail effectué

J'ai remarqué que lors du scan en mode mémoire l'exclusion par appui long sur M ne fonctionne que si la réception est active, si la réception s'arrête et que l'on appui pour exclure la mémoire pendant le laps de temps que met le scan à reprendre on ne peut pas exclure la mémoire... serait il possible de modifier le fonctionnement ?

d'avance merci et bonnes fêtes de fin d'année

Stéphane

armel, 2024-12-27:
Bonsoir Stéphane,

Je ne vois pas bien l'intérêt à exclure un canal mémoire sur lequel il n'y a aucune émission en cours ?
Quel est le cas d'usage ?

Armel F4HWN.

stephane-rib, 2024-12-28:
si par exemple je veux exclure un canal pmr sur lequel il y a un échange régulier entres des personnes et que je veux écouter seulement les autres canaux sans que le scan s'arrête régulièrement sur celui ci

MrHoerby, 2024-12-28:
https://github.com/armel/uv-k5-firmware-custom/wiki/Button-functions#5-noaa

closed 2025-01-01 by armel