armel/uv-k5-firmware-custom #353: Verrouillage volume ...

Daffouille, 2025-01-10:
Bonne et heureuse année 2025 à toi, et encore merci pour ton travail sur ce FW

J'ai peut être raté quelque chose, mais même si je verrouille le poste (clavier) et que je glisse le poste dans ma poche, le bouton volume reste actif ... et là, advienne que pourra ...

Je sais, ce n'est peut être pas le "use case" du siècle, mais toutes les remarques peuvent être utiles ...

Bref je te pousse le sujet, vois ce que tu veux/peux faire ....

73

armel, 2025-01-11:
Bonjour,

Merci pour ton message et meilleurs vœux à toi, ainsi qu'à tes proches.

Concernant ta question, le problème c'est qu'il n'y a **aucun** moyen de contrôler l'état du bouton rotatif du K5. 
C'est juste un bouton de marche / arrêt, puis de réglage du volume.
Mais aucun moyen de le verrouiller ou de savoir si tu l'as tourné d'un quart de tour par exemple, par voie logiciel.

Je ne peux donc rien faire ici, autrement que te donner le truc que j'utilise quand je suis en randonnée et que le K5 est (parfois) rangé dans une poche (par exemple, s'il pleut) : je retire tout simplement le bouton. Ca vaut ce que ca vaut, mais ca fait le job.

Armel, F4HWN.

fcatt, 2025-01-11:
Haha je compatis ça m'arrive aussi :smile: 
Tu peux aussi changer le bouton pour un autre moins proéminent, par exemple celui-là imprimé en 3D:
https://www.thingiverse.com/thing:6163862
Ou ajouter un anneau de protection:
https://www.printables.com/model/704894-volume-knob-protection-quansheng-uv-k5-uv-k6/

armel, 2025-01-11:
@fcatt Oui, l'impression 3D est aussi une solution. 

[image]

Daffouille, 2025-01-11:
Bonjour
Merci à tous pour vos idées !

closed 2025-01-12 by armel