armel/uv-k5-firmware-custom #354: Compiling the firmware with a local docker fails with a landlock error

markusb, 2025-01-13:
When I attempt to build the firmware using the compile-with-docker.sh script fails with the following error:

```
$ ./compile-with-docker.sh voxless
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  105.5MB
Step 1/10 : FROM --platform=amd64 archlinux:base
base: Pulling from library/archlinux
Digest: sha256:292bb4ec7866af67f8d66fe2114eef840eebc74302db00b891dcb2d278bec80a
Status: Downloaded newer image for archlinux:base
 ---> cdf34a96768a
Step 2/10 : RUN pacman -Syyu base-devel --noconfirm
 ---> Running in 47824ce48e80
:: Synchronizing package databases...
 core downloading...
 extra downloading...
error: restricting filesystem access failed because the landlock ruleset could not be applied!
error: failed retrieving file 'extra.db' from geo.mirror.pkgbuild.com : Could not resolve host: geo.mirror.pkgbuild.com
warning: fatal error from geo.mirror.pkgbuild.com, skipping for the remainder of this transaction
error: failed retrieving file 'core.db' from geo.mirror.pkgbuild.com : Could not resolve host: geo.mirror.pkgbuild.com
error: failed retrieving file 'extra.db' from mirror.rackspace.com : Could not resolve host: mirror.rackspace.com
warning: fatal error from mirror.rackspace.com, skipping for the remainder of this transaction
error: failed retrieving file 'core.db' from mirror.rackspace.com : Could not resolve host: mirror.rackspace.com
error: failed retrieving file 'extra.db' from mirror.leaseweb.net : Could not resolve host: mirror.leaseweb.net
warning: fatal error from mirror.leaseweb.net, skipping for the remainder of this transaction
error: failed retrieving file 'core.db' from mirror.leaseweb.net : Could not resolve host: mirror.leaseweb.net
error: failed to synchronize all databases (failed to retrieve some files)
The command '/bin/sh -c pacman -Syyu base-devel --noconfirm' returned a non-zero code: 1
Failed to build docker image
```

From my research this seems to be because sofe some landlock configuration issue in the arch linux base image. I did attempt to disable landlock by adding `ENV LANDLOCK_DISABLE=1` to my Dockerfile. Unfortunately with no change in result.

armel, 2025-01-13:
I have no idea. 
I see this message several times in your log -> _Could not resolve host_
It's not a problem with your DNS resolver ?

It works like a charm for me under MacOS.

Armel, F4HWN.

zardoz99, 2025-01-14:
Landlock is not the issue. It is the host resolution that is your problem. The landlock message appears on my Intel based, Fedora 41, installation and the compile still completes successfully.

armel, 2025-01-14:
> Landlock is not the issue. It is the host resolution that is your problem. The landlock message appears on my Intel based, Fedora 41, installation and the compile still completes successfully.

Thank you @zardoz99. That's my opinion too. 

Armel, F4HWN

markusb, 2025-01-14:
Host resolution is not the problem. The host can be resolved with no issue:
```
(venv) markus@altair:~/Workbench/Quansheng-UV-K5/F4HWN-uv-k5-firmware-custom$ ping geo.mirror.pkgbuild.com
PING geo.mirror.pkgbuild.com (2a02:6ea0:c237::2) 56 data bytes
64 bytes from 2a02:6ea0:c237::2: icmp_seq=1 ttl=54 time=22.0 ms
64 bytes from 2a02:6ea0:c237::2: icmp_seq=2 ttl=54 time=21.5 ms
64 bytes from 2a02:6ea0:c237::2: icmp_seq=3 ttl=54 time=21.5 ms
^C
--- geo.mirror.pkgbuild.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 21.488/21.660/22.003/0.242 ms

```
On the same system I have other containers in good working order who are able to reach the internet (home assistant).

prokrypt, 2025-01-14:
sure seems like a host resolution problem to me ¯\\\_(ツ)\_/¯. go into that container to check.

markusb, 2025-01-17:
I found the problem. The `docker build` command in the `compile-with-docker.sh` script is incomplete. 

The current command `docker build -t $IMAGE_NAME` is missing a reference, so no network access is possible.

If I replace this with `docker build --network=host -t $IMAGE_NAME .` then it works.

Can we add a sentence about the requirement to adding `--network=<network>` to the `docker build` command in the README?

armel, 2025-01-17:
Ok @markusb.

In my opinion, the use of `--network=host` indicates an issue specific to your network environment (likely a proxy or a firewall). While this option may quickly resolve your problem, it introduces **security risks** by removing network isolation. Therefore, I don’t think it’s a good idea to include this tip in the README. By the way, you’re the first (and so far the only one) to encounter this issue.

Armel, F4HWN.

armel, 2025-01-19:
@markusb PR #355 merged on [feature_update_v4](https://github.com/armel/uv-k5-firmware-custom/tree/feature_update_v4). Thank you.

closed 2025-01-19 by armel