armel/uv-k5-firmware-custom #360: Refresh screen to solve screen rubbisch after HF interference.

ErikS-web, 2025-02-05:
Seems a solution for RFI interference on screen of UVK5(8) with orange screen is not present or working in firmware 3.9

You firmware is a fork of Egzumer, where the firmware workaround was present.

Could you check ?

[image]

**reference.**
https://github.com/egzumer/uv-k5-firmware-custom/issues/60

_some text from that issue ..._

This [enhancement] might be a solution for UV-K5(8) where the orange backlighted screen is sensitive for HF.
This sometimes rusults in stripes on screen or blanc screen.
Just place hand-palm on screen, with fingers near antenna, and transmit. And screen becomes rubbisch.

It seems that RFI is interfering with display I2C communication when display contents are changed. Indeed, it's a hardware thing, but perhaps it could be worked around in SW.

Solved with "screen initialization" after each PTT-release. in EGZUMER [Release 0.16](https://github.com/egzumer/uv-k5-firmware-custom/releases/tag/v0.16)

ErikS-web, 2025-02-05:
video https://t.me/quanshenguv5kde/23675

and discussion in DE-telegram-group

but 

we just checked the Source code, the fix from egzumer is present in F4WHN.

[image]

https://t.me/quanshenguv5kde/23772

armel, 2025-02-05:
Hello,

My firmware is a fork of [Egzumer 0.22](https://github.com/egzumer/uv-k5-firmware-custom/releases/tag/v0.22).

Armel, F4HWN

ErikS-web, 2025-02-05:
EGZUMER 0.22 and Nunu work fine. But your 3.9 not.
Wil test ....

Just flashed EGZUMER 0.22 on my F4HWN-3.9-radio and fix work fine and restores screen with 0.22

also Nunu 20.5

Back2 [Bandscope version (without FM broadcast support)](https://egzumer.github.io/uvtools/?firmwareURL=https://github.com/armel/uv-k5-firmware-custom/raw/main/archive/f4hwn.bandscope.packed.v3.9.bin)
and problem is back.

So yes. there seems to be a bug.

sparky93a, 2025-02-05:
stock antennas, multiple radios all do it on 70cm too

more severe than  prior times. use of menu does not refresh(before egzumer fix) only way to refresh is a power cycle.sometimes even see a distoreted clock 02:43.00 hh mm ss. with a flashing :

ErikS-web, 2025-02-05:
Now after reflashing I can not reproduce the error and screen restores normal after RFI.

Now I also restore setting with your Chirp.py and problem is back,,,,

Oeps ... I do not use the latest Chirp driver v3.9.1 but 3.8.0.

ErikS-web, 2025-02-05:
Now with 3.9.1. chirp driver settings reloaded and problem solved !

Wil do some testing tomorrow, before closing this issue

armel, 2025-02-05:
Please be careful. You seem to change firmware often. Note that my firmware does not modify the eeprom and calibration. Neither does Egzumer. But I don't know if this is the case with Nunu (firmware I don't know). I can only suggest that you cleanly restore the eeprom with an original backup BEFORE installing my firmware on your transceiver, especially if you have tested another firmware before (IJV, Fagci, may be Nune, etc.). Otherwise, I can't vouch for anything...

Again, I've been developing this firmware for 14 months. I understand that it is widely used around the world. And apart from HF return problems linked to excessive SWR (I've even made [a video](https://www.youtube.com/watch?v=nLmCYL315yA&ab_channel=F4HWNArmel) on the subject in French...) due to **misuse**, no problems have been reported. 

Armel, F4HWN.

armel, 2025-02-06:
> problem returns with reloading my Chirp-backup

Even if Nunu is an Egzumer fork, are you sure it hasn't updated the calibration data? 

> So far this problem seems solved after Reset ALL (extended menu)

Yes, RESET ALL can be useful.
But has I said, I suggest you to cleanly restore the eeprom with an original (stock) backup BEFORE installing my (or any other) firmware on your transceiver. It's always best to start with a clean (stock) eeprom, to avoid any surprises. 

> I test om 144.6 Mhz High power FM with QS-antenna

Again, I'm pretty sure your issue is related to SWR, not the firmware. As I explained in the [video](https://youtu.be/nLmCYL315yA?t=370), try testing with a 50-ohm dummy load and see for yourself.

Armel, F4HWN.

ErikS-web, 2025-02-08:
You have pasted the label INVALID. 

But I think I found the trigger and will describe it to you. And I hope it helps you recognize what it is and adjust the screen reinitialization. 
It only occurs with your firmware. Not with EGZUMER, NUNU and Q-dock that I test on separate radios.

I will explain the steps.

ErikS-web, 2025-02-08:
First **reset ALL** in _Hidden menu_. After that, the problem of a persistent RFI **_no longer occurs_** (Display disturbance after releasing PTT). 
I was very surprised and started investigating further....

When I reload my backup (the chirp setting) I have the RFI error after PTT again.

Then I do Hidden-menu **Reset VFO.** Then the problem of a persistent RFI is solved.
**_So it depends on the channel list and not on settings._**

When I save a 2 meter frequency like 144.600 in **memory** and select that Memory channel on screen the RFI-problem is back. 

**With 2 vfo channels on screen NO problem.** 

Same problem on MAIN-only with memory channel selected (one VFO on screen)

_**Possible cause**_

Only thing what I think is extra text on the screen is scanlist 123. in stead if I or II (what is the Egzumer standard)

**_Solution proposal_**

My suspision it that reinit-screen-routine (after PTT-release) does not handle this "scanlist 123". 
And display is not rewritten correctly, to RFI problem stuys on screen after PTT-release.

ErikS-web, 2025-02-08:
I just saw the video you made.
https://www.youtube.com/watch?v=nLmCYL315yA&t=370s

I don't have that beautiful blue radio yet and I still have to order it.
But it does seem to be more sensitive than the UVK5(8).
You do indeed explain nicely why. Compliments.

But look where VFO-B is on... a **_memory channel (M18)_**

[image]

Do the test again and then without a memory channel in the VFO ;-)

armel, 2025-02-08:
> But look where VFO-B is on... a memory channel (M18)
> Do the test again and then without a memory channel in the VFO ;-)

Observe the position of the arrow. The VHF frequency is selected. Repeating the test, then without a memory channel in the VFO, will not change anything. 

Armel, F4HWN.

armel, 2025-02-08:
> Solution proposal
>
> My suspision it that reinit-screen-routine (after PTT-release) does not handle this "scanlist 123".
> And display is not rewritten correctly, to RFI problem stays on screen after PTT-release.

Here's a suggestion: send me a complete **dump of your eeprom** that I can inject into one of my K5 (8) (the same as you...). And describe to me, step by step, how to simply reproduce your problem at any time. And I'll try to do it with default antenna and with a dummy load... 

I can enter the same frequency as you (144.600 MHz), use VFO or memory channel, select HIGH power and transmit, without having the problem. **At least, as long as I have the transceiver in hand (because of the normal SWR problem)**. By the way, I've been doing QSOs (on another frequency, 145.500 MHz) every day for over a year. 

I'm waiting to hear from you, about your eeprom dump and the procedure. 

Armel, F4HWN

ErikS-web, 2025-02-08:
You can do your test right now.

Just as shown in the video, but on VBO-B do not have a Memory-channnel active, but a VFO-frequency

Can I contact you via QRZ ?

armel, 2025-02-08:
> You can do your test right now.
>
> Just as shown in the video, but on VBO-B do not have a Memory-channnel active, but a VFO-frequency

It works like a charm.

> Can I contact you via QRZ ?

Yes, of course.

Armel, F4HWN

armel, 2025-02-10:
Hi,

Thank you for sending me your eeprom. As expected, everything is working normally. 

https://github.com/user-attachments/assets/2bf2367c-635f-41c1-a650-4dc576c3c17c

Armel, F4HWN

ErikS-web, 2025-02-10:
... try harder ;-)

Suggest you _**use 3 fingers**_, like in my photo ;-)

Thumb on the screw
Middle finger on the antenna coil
and _**move the Index finger**_ across the screen.

[image]

- Just tried another radio, from a local radio amateur. Had Egzumer on it.

Did Reset-All after flashing your firmware and now only 144,600 in memory.

Here starting picture. 

[image]

It does NOT matter if Memory is on A or B and what is Main channel A or B

His radio seems somewhat less sensitive...
I had to try more than 10 times....and I now have 2xRFI screen 

[image]

armel, 2025-02-10:
> When external antenna (on roof) is used I never have RFI interference on the screen. Did not test with by replacing the antenna with a 50 ohm dummy load. But espect that same result.

Indeed, the result will ne the same. 
You can also redo your test at LOW1 power (<20mW), under your current conditions (stock antenna). And the problem won't be there, as the HF feedback is insignificant.

Note that, compared to Egzumer, power HIGH is a little more increase avec mon firmware. 

But keep in mind, once again, that using this transceiver on a table, One Push activated, without counterpoise, at HIGH power is nonsense. You'd have the same problems, more or less quickly, with an Icom ID-52 (I did it...).

And finally, this is not to the software (i.e my firmware) to deal with this kind of hardware issue.

So what ? 

Armel, F4HWN

armel, 2025-02-10:
Last but not least, persisting in using your transceiver under these conditions (OnePush, HIGH Power, without counterpoise, etc.) will lead to destruct it.

Armel, F4HWN.

ErikS-web, 2025-02-10:
There is a Quick hardware fix ...

[image]

-  Ground metal plate of display. I this example with ground of antenna.

armel, 2025-02-10:
So, 24 messages to arrive, at the end, at a problem of ROS and counterpoise. 
As I said from the beginning. 
Doing Open Source is sometimes a real challenge...
I think it's time to close this (non) issue. 

Armel, F4HWN.

closed 2025-02-10 by armel