armel/uv-k5-firmware-custom #363: Chirp driver not working as expected

rbzikadze, 2025-02-09:
In version 3.9, the provided CHIRP driver does not function correctly when reading from or uploading to the radio. When attempting to read memory, the list appears empty. Additionally, after creating a new memory channel in CHIRP and uploading it to the radio, the memory does not appear.

I have also tested the CHIRP driver version 3.9.1 available [here](https://github.com/armel/uv-k5-chirp-driver), and encountered the same issue.

Attempted reflashing the firmware
Reset VFO and performed a full reset
Reflashed the firmware again
Despite these steps, the issue persists. Any insights or suggestions would be appreciated.

armel, 2025-02-09:
First of all, Hi...

Why did you post a problem on the firmware repository, if it's a problem with the [Chirp driver](https://github.com/armel/uv-k5-chirp-driver)?

Believe me, the Chirp driver works like a charm. By the way, there is no open issue about it.

Has your K5 been modified? Have you replaced the eeprom with a new, larger one?

I'll leave this issue open, if other people want to confirm or deny what I've said.  

Armel, F4HWN

armel, 2025-02-09:
Yes, of course. Some K5 users have upgraded their eeprom to one with a larger capacity. The default eeprom is 8 Kb. They replace it with a 64 Kb eeprom. If the firmware they're using supports it (e.g. IJV), this means they can store 999 memory channels, for example. 

Coming back to your problem, I don't know Kamils' firmware, sorry. But it's possible that it changes the way data is stored in the eeprom. My firmware, like Egzumer's, does not change the way data is stored. 

Did you have the good idea of making a backup of your eeprom (out of the box) BEFORE installing other firmwares, like Kamils'? If so, restore your eeprom first and then install my firmware. 

Armel, F4HWN.

rbzikadze, 2025-02-10:
I have the original eeprom taken before flashing any of the firmwares. I have restored the original eeprom, reflashed original firmware, then flashed f4hwn v3.9 and noticed that after eeprom read with k5prog it comes up empty, writing eeprom with k5prog also results in nothing (channels and settings don't change on device), reading eeprom again after eeprom writ still results in an empty read. 

Also i've noticed a  minor inconsistency in spectrum analyzer, * and F buttons are reversed, * should move squelch line up and F should move it down, makes less sense the way it is now

armel, 2025-02-10:
> I have the original eeprom taken before flashing any of the firmwares. I have restored the original eeprom, reflashed original firmware, then flashed f4hwn v3.9 and noticed that after eeprom read [...]

Sorry, I'm not sure I understand. Does this mean that even with k5prog, you can't write / modify the eeprom?

> Also i've noticed a minor inconsistency in spectrum analyzer, * and F buttons are reversed, * should move squelch line up and F should move it down, makes less sense the way it is now

And you're going to tell me that moving around the menu with the up and down arrows isn't logical either? 😉

Armel, F4HWN.

rbzikadze, 2025-02-10:
Yes k5prog doesn't read/write eeprom with firmware 3.9, have tried multiple different firmwares and they don't have issues with reading or writing eeprom. Up/Down is logical, * should be up  and F should be down not the other way around imho. When was the last time you have tried and tested chirp and/or k5prog to read eeprom?

armel, 2025-02-10:
I don't understand why you can't read/write to the eeprom. Again, it's not link to the firmware.

https://github.com/user-attachments/assets/a8bacd33-3a7f-4fc6-8a4b-03cf11607e2d

So 2 solutions:

- Maybe ask for assistance on the [Facebook group](https://www.facebook.com/groups/6780091945388779) I've indicated. Maybe someone other than me will have an idea,
- Continue to use Egzumer, Kamils or stock firmware.

Once again, my firmware and the Chirp driver are used by thousands of users all over the world, every day. 
I can't solve a problem I can't reproduce. I'm not a magician, sorry.

I'm going to leave this issue open for a while, if other people know how to help you. I don't.

Armel, F4HWN

rbzikadze, 2025-02-10:
https://github.com/user-attachments/assets/047cf2c7-903f-45aa-b7b0-83a97901b7c0
https://github.com/user-attachments/assets/dc2e9314-e37f-4a0b-8514-59064b7f1e28

Here is the demonstration with other [firmware](https://github.com/kamilsss655/uv-k5-firmware-custom) and your latest 3.9 firmware when using k5tool to read eeprom, notice that read goes without errors for both but the resulting file is empty on f4hwn.

rbzikadze, 2025-02-10:
Here is the hexdump of eeprom file after running ./k5tool -rdee k5_eeprom.raw 
[image]

armel, 2025-02-10:
Can you try this with a K5 **with F4HWN firmware**?

1 - `./k5tool -hello`
2 - `./k5tool -wree k5_eeprom_kamils.raw`
3 - `./k5tool -rdee test.raw`
4 - `hexdump -C test.raw`

And send me screenshot, pls.
By the way, do you have a second K5 to test on another transceiver?

Armel, F4HWN.

rbzikadze, 2025-02-10:
Here's the reproduction
[image]

rbzikadze, 2025-02-10:
I do have another transceiver and will try to reproduce on it soone and will do an update

rbzikadze, 2025-02-10:
Exactly same behavior on my other radio

armel, 2025-02-10:
Hey... the real question is : why are you using a HasCustomAesKey !
First time I see this...

[image]

Armel, F4HWN.

rbzikadze, 2025-02-10:
I am actually not sure sure about that value or how it could be set to 1, where can I find more info about that?

rbzikadze, 2025-02-10:
I've found this in firmware maybe this has something to do with HasCustomAesKey?

https://github.com/armel/uv-k5-firmware-custom/blob/d365896abb989dfa729463c1c000a6ee59b5f733/settings.c#L336

rbzikadze, 2025-02-10:
I see, there is an encryption key that is stored there for messaging functionality in kamils firmware, but I've tried flashing original eeprom which is empty in 0F30..0F3F then reflashed v3.9 and still I get an empty eeprom read. Doesn't matter if encryption key is there or not in eeprom address 00000f30 while v3.9 is installed I still get empty reads

armel, 2025-02-10:
> flashing original eeprom which is empty in 0F30..0F3F then reflashed v3.9 and still I get an empty eeprom read.

Empty or with 0xff ???
K5tool -hello said always HasCustomAesKey: 1 after this ?

Seriously, I can'f fix problems because of other firmware usage. 
My firmware doesn't deal with custom AES because nobody use it and this feature take lot's of memory. 

Armel, F4HWN.

rbzikadze, 2025-02-10:
With stock eeprom k5tool says HasCustomAesKey: 0. With this eeprom on device I flash v3.9 firmware, then also k5tool -hello says HasCustomAesKey:0, I try to read eeprom and result is empty eeprom file, no bytes in it at all so this has nothing to do with HasCustomAesKey is 1 or 0

armel, 2025-02-10:
I don't think so.
Could you flash firmware [F4HWN v3.2 ](https://github.com/armel/uv-k5-firmware-custom/releases/tag/v3.2) and retry with this old version ?

Armel, F4HWN.

armel, 2025-02-10:
By the way, could you send me your k5_eeprom_kamils.raw dump (zipped format) in attachement ?

Armel, F4HWN.

rbzikadze, 2025-02-10:
[kamils_radio1.zip](https://github.com/user-attachments/files/18741114/kamils_radio1.zip)

armel, 2025-02-10:
Thank you, I'll check.

Armel, F4HWN.

rbzikadze, 2025-02-10:
By the way test1234 is the encryption key for messaging app in that eeprom

armel, 2025-02-10:
I know :) 
I've got an urgent job to finish, but believe me, I can already tell you that the problem is indeed with your custom AES key, as I said...
So stay tuned...

Armel, F4HWN

armel, 2025-02-10:
Unzip and flash this firmware v3.9 and tell me.

[f4hwn.bandscope.packed.bin.zip](https://github.com/user-attachments/files/18741904/f4hwn.bandscope.packed.bin.zip)

Armel, F4HWN

rbzikadze, 2025-02-10:
I've flashed this and it works, I can read the eeprom with k5tool and also can read memory with chirp. This is awesome, thanks so much. What was the issue and a fix?, I'm interested

armel, 2025-02-11:
As I said, my firmware doesn't support the custom AES key because nobody uses it and it takes up a lot of memory. In fact, you're the first (and probably only?) user to have tried to use my firmware with a personalized AES key, since last year. 

So I've decided to definitely bypass this custom AES key if it exists. 

I think it's time to close this issue.

Armel, F4HWN

closed 2025-02-11 by armel

armel, 2025-02-11:
Thank you @F4ICR 😌

darkymtp, 2025-02-16:
Hello.
I have the same behaviour, AES key is set albeit I've never used that feature.
Your modified 3.9 is working.

dv3ppe, 2025-02-24:
i got the same issue as yours..i solved the problem by these..TAKE NOTE BY THE SEQUENCE!!!
1. by using k5prog, upload IJV v3.4 then full reset.
2. upload f4hwn v3.7 or v3.8 not direct v3.9..then full reset..
3. problem solve..i think these is the issue of refactoring..
4. thanks to armel..great stuff..waiting for tk11..hopes this helps..73

dv3ppe, 2025-02-24:
i test many firmwares that could benefit me and my group..armel's fw is one i've use as my daily driver..take note!!!..using armel's fw, you need to try the previous versions before upgrading to the latest one..trial and error..goodluck and 73

dv3ppe, 2025-02-24:
to armel,

i hope you could also integrate the bandscope while in memory mode just like kamilsss655..we would greatly appreciate it..more faster scanning on memory mode..tnx

armel, 2025-02-24:
> to armel,
> 
> i hope you could also integrate the bandscope while in memory mode just like kamilsss655..we would greatly appreciate it..more faster scanning on memory mode..tnx

I think it will be complicated, given the limited memory available. But I'll take the idea, my friend. 

Armel, F4HWN

dv3ppe, 2025-02-24:
Yeah I agree..maybe i just missed the qso's haha..take care