armel/uv-k5-firmware-custom #370: Inversion entre TxCTCSS et RxCTSS

F6CYK, 2025-02-25:
Bonjour Armel,

En mode VFO, il semblerait qu'il y ait une inversion entre le menu TxCTSS (06/67) et RxCTSS (04/67). À titre d'exemple, si l'on programme à la volée le TxCTSS (06/67) sur 114,8 Hz et le RxCTSS (04/67) sur 167,9 Hz le passage en émission envoie 167,9 Hz et la réception se fait sur 114,8 Hz. L'ndication est par contre correcte sur l'écran à l'exception de la lettre "R" qui reste en émission et en réception. 

73 de F6CYK et encore bravo pour le travail effectué au profit de la communauté.

armel, 2025-02-25:
Tu es en reverse... donc forcément, tout est inversé.
Fais un appui long sur la touche 8 R pour sortir de ce mode.
Et tout rentrera dans l'ordre.

73'

Armel, F4HWN.

F6CYK, 2025-02-25:
En effet, c'est stupide de ma part :-D Je me suis fait bêtement avoir en travaillant sur trois unités en même temps et comme j'étais sur la même fréquence en émission et en réception, je ne me suis pas aperçu que j'étais passé en "reverse" :-/
Merci et 73 de F6CYK

armel, 2025-02-25:
Pas de problème, Pierre-Antoine.
Ca peut arriver à tout le monde.
Comme je le dis souvent, ce petit poste est si polyvalent qu'on finit parfois par s’y perdre ! 😊
Je cloture l'issue.

Excellente soirée à toi et 73'

Armel, F4HWN

closed 2025-02-25 by armel