armel/uv-k5-firmware-custom #372: v4.0 bandscope edition

Daffouille, 2025-03-04:
Bonsoir Armel

Je ne sais pas s'il s'agit d'une mauvaise manip/incomprehension de ma part, mais avec un SetOff à 1 minute, le poste passe en veille tout seul lorsque je suis en Scan (et sans recevoir une quelconque émission sur les mémoires scannées).

Je m'attendais à ce qu'il n'y passe que lorsqu'il était volontairement arrêté sur une fréquence de mon choix, et pas en balaye volontaire des canaux mémoire (d'une liste).

Bonne soirée.

armel, 2025-03-04:
Bonsoir,

C'est le comportement normal.

Conformément à la [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#about-the-setoff-menu), si:

- pas de réception,
- pas d'émission, 
- pas d'appui sur une touche.

Le poste se met en veille une fois le delay SetOff atteint. 
Je précise, en complément et au cas ou, que **le scan continu en tache de fond.**..il sera juste plus ou moins réactif, selon la valeur du paramètre BatSav que tu auras choisi.

73'

Armel, F4HWN.

ldicdican, 2025-03-05:
Good day Armel,

May I add my experience here with the current firmware 4.0, when I'm scanning on VFO or channel memories and didn't stop the scanning and turn the device off, then when you turn it on again you will see that the scanning is not stopped, it automatically resumes its scanning at the time you turned the device off. Is this normal?

armel, 2025-03-05:
> Good day Armel,

Hello,
 
> May I add my experience here with the current firmware 4.0, when I'm scanning on VFO or channel memories and didn't stop the scanning and turn the device off, then when you turn it on again you will [...]

It's not a bug, it's a feature 👍 

Please, take a look to the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#scan-resume-%EF%B8%8F-disable-in-basic-edition).

Armel, F4HWN

ldicdican, 2025-03-05:
Thank you Armel for reminding me about its documentation, will read it thoroughly. Great firmware and continuously appreciated using it 👍👍

armel, 2025-03-05:
> Thank you Armel for reminding me about its documentation, will read it thoroughly. Great firmware and continuously appreciated using it 👍👍

Thank you sir !

Armel, F4HWN

closed 2025-03-06 by armel