armel/uv-k5-firmware-custom #392: VOX and SQL are mutually exclusive

s3c, 2025-04-24:
Unlike the stock firmware, you cannot use VOX and SQL - 0 at the same time. This makes APRS less reliable, since you need VOX and to be in monitor mode. You thus have to use SQL != 0, and wait for the SQL to open, which misses the beginning of your incoming packet. I currently get around this by setting a longer preamble on my digipeater, but that's not possible for other digipeaters. Ideally you would want to leave the radio with the SQL open, and use VOX to trigger TX.

armel, 2025-04-25:
First of all, hello,

To be honest, I'm not quite sure how to respond. I'm not entirely clear on your use case, and I'm not an expert in APRS or digipeaters.

From my perspective, the VOX feature is hardly usable—even in the stock firmware. I have a feeling that 99% of users never use it at all. In fact, I'm even considering removing it in a future release if I run out of space.

So as it stands, I don't think there's much I can do on this front.

Armel, F4HWN

hackra76, 2025-04-28:
Hi armel, just a quick question. 
Would it be possible to remove the VOX function (which I too think is completely useless) and add the FM broadcast to the Bandscope version ? I love the bandscope, but I miss the FM broadcast. When hiking, I often like to listen to a radio station, but now I have to carry another device with me. I have no idea how much memory any of those functions require, so please excuse me if this is a stupid question. Thank you.

hackra76, 2025-04-28:
> Its already available... Look at basic addition!!
> 
> [image]

Yes I know, but the Basic edition is missing several features I find useful, so my idea was to make a Bandscope+FM broadcast edition without VOX feature. I understand that is probably because removing just the VOX feature is not enough free memory to enable both Bandscope and FM broadcast. Or am I wrong ?

hackra76, 2025-04-28:
Maybe I didn't write it correctly, as English is not my native language, What I meant by my first comment was, I would like to have a Bandscope version with all its perks and whistles + FM broadcast feature, but without the VOX nonsense.... If it is impossible, I am OK with it, just say so. But if it is possible, I would be highly interested. That is all.

armel, 2025-04-28:
Hello,

As @lohtse rightly said, no, it's not possible. There simply isn't enough memory available.
If it had been possible, I wouldn't have proposed different editions.
So you have to make some compromises and choose the edition that best matches your needs.
Another option: own several K5 units and install the different editions.

I am closing this issue.

Armel, F4HWN

closed 2025-04-28 by armel